crayzeewulf/libserial #13: Issue while using  in a process with multiple threads

crayzeewulf, 2015-06-03:
**Reported by nikhilmahale on 2012-02-04 14:09 UTC**
At the outset I would like to state that I am relatively new to libserial, C++ programming and ROS, so please excuse any obvious mistakes.
I am using libserial to communicate with peripherals, in applications developed using the ROS framework \(http://www.ros.org/wiki\). I was experiencing read timeouts \(even for large values of timeouts\), random wrong data being read from the port, std::runtime\_error 's in read and write operations to ports and segmentation faults  in random sections of code when using libserial as a part of ROS Nodes \(applications\). The same programs worked fine when I did not use any ROS components in them. Similarly programs using ROS framework without  libserial worked fine. 
The only difference I could find was that, programs using ROS components caused a process to spawn multiple threads while without ROS the process executed a single thread. 
After some debugging I found that replacing the following code in &lt;SerialPort::SerialPortImpl::Open&gt;
/\*
\* Direct all SIGIO and SIGURG signals for the port to the current
\* process.
\*/
if \( fcntl\( mFileDescriptor,
F\_SETOWN,
getpid\(\) \) &lt; 0 \)
\{
throw SerialPort::OpenFailed\( strerror\(errno\) \) ;
\}

with 

/\*
\* Direct all SIGIO and SIGURG signals for the port to the current
\* thread.
\*/
struct f\_owner\_ex this\_thread;
this\_thread.type = F\_OWNER\_TID;
this\_thread.pid = syscall\(SYS\_gettid\);
if \( fcntl\( mFileDescriptor,
F\_SETOWN\_EX,
&this\_thread \) &lt; 0 \)
\{
throw SerialPort::OpenFailed\( strerror\(errno\) \) ;
\}

has solved my problems. I have tested the applications thoroughly. Before applications would crash after running for anywhere from a couple of minutes to some hours. Now they run continuously without any of the above stated errors for as long as required.

mcsauder, 2017-09-13:
Hi Crayzee Wulf and nikhilmahale,

I have previously reproduced the multi-thread crash occurrences for both SerialPort and SerialStream class objects in the master branch by utilizing the [mutli-thread unit tests in the c++14 branch](https://github.com/crayzeewulf/libserial/blob/39bdf19806e8448c0f0baef7a29500e261af6826/test/LibSerialTests.cpp#L1529).  The unit tests also serve to validate the fixes in place in the [c++14 branch](https://github.com/crayzeewulf/libserial/tree/c%2B%2B14).  Let me know if you have any questions!

-Mark

mcsauder, 2018-02-08:
Fixed with Pull Request #102!

closed 2018-02-08 by mcsauder