crayzeewulf/libserial #53: no configure

RAtechntukan, 2015-06-10:
configure script isn't generate

crayzeewulf, 2015-06-10:
Could you please provide more information such as:

* What did you try to generate the `configure` script?
* Was there any error message when you tried to generate the `configure` script?
* What operating system and version as you using to build `libserial`?
* Did you try running `autoreconf`? If so, what happened? Were there any error messages? What command line options did you use?
* Did you try running `make -f Makefile.dist`? If so, what happened? Were there any error messages?

crayzeewulf, 2015-06-10:
RAtechntukan, could you please try again using the latest version in git and let me know if it solves your problem? The issue could have been a missing `README` file. But if it still does not work, please answer some of my questions in the previous comment.

tdack, 2015-06-28:
I had the same problem after grabbing master from git.

I had to do the following:
* install autoconf
* install libtool

Then `make -f Makefile.dist` before running `./configure`

It also looks like libserial depends on part of the boost libraries

Radioguy00, 2015-07-02:
I previously installed this library on a different machine and I did not have any problem. It looks like now the configure script has disapeared from the list of files under version control?

crayzeewulf, 2015-07-06:
All good points. The `configure` script has never been under revision control because it is generated by GNU autotools. When the library code was distributed on SourceForce, the tarball was created by running `make dist`. This bundles the auto-generated `configure` script in the `tar.gz` file it creates. As expected, the code in the git repository does not have the `configure` script either. I have updated the top-level `README.md` file to indicate that you will have to run `make -f Makefile.dist` in order to generate the `configure` script. This is similar to running `autogen.sh` scripts that are sometimes provided by other packages (for example, see [google/protobuf](https://github.com/google/protobuf)).

closed 2015-08-02 by crayzeewulf