crayzeewulf/libserial #54: Release for 0.6.0rc2 and rc3

joerg-krause, 2015-07-30:
Can you provide a release tag for 0.6.0rc2 and rc3, please?

crayzeewulf, 2015-08-02:
I have created a tag for 0.6.0rc2 corresponding to the code that was packaged on SourceForge. As far as I can tell, there never was a 0.6.0rc3 release; the number was incremented in `configure.in` in preparation of the next release candidate. Hope this helps. Let me know if you need anything else.

joerg-krause, 2015-08-02:
I see! So rc3 is not released, yet.

Will new release only be available on github or is the sourceforge repo still maintained?

crayzeewulf, 2015-08-03:
The sourceforge repository is no longer being maintained and new releases will only be made on github in the future. The current plan is to remove the sourceforge repository and project sometime in the near future.

closed 2015-08-03 by joerg-krause