crayzeewulf/libserial #58: Example program won't run; undefined debug symbol

JohnBabrick, 2016-01-25:
I am building in eclipse the receive example program and get this error when I try to run it:

/home/johann/workspace/testSerial/Debug/testSerial: symbol lookup error: /home/johann/workspace/testSerial/Debug/testSerial: undefined symbol: _ZN9LibSerial12SerialStream4OpenENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt13_Ios_Openmode

I have added the -lserial and the -pthread and the appropriate lib and include directories. I built the code from source on the latest from github under Ubuntu 15.04. Any ideas? 

#include <SerialStream.h>
#include <iostream>
#include <unistd.h>
#include <cstdlib>

int
main( int    argc,
      char** argv  )
{
    //
    // Open the serial port.
    //
    using namespace LibSerial ;
    SerialStream serial_port ;
    serial_port.Open( "/dev/ttyUSB0" ) ;
    if ( ! serial_port.good() )
    {
        std::cerr << "[" << __FILE__ << ":" << __LINE__ << "] "
                  << "Error: Could not open serial port."
                  << std::endl ;
        exit(1) ;
    }
    //
    // Set the baud rate of the serial port.
    //
    serial_port.SetBaudRate( SerialStreamBuf::BAUD_115200 ) ;
    if ( ! serial_port.good() )
    {
        std::cerr << "Error: Could not set the baud rate." << std::endl ;
        exit(1) ;
    }
    //
    // Set the number of data bits.
    //
    serial_port.SetCharSize( SerialStreamBuf::CHAR_SIZE_8 ) ;
    if ( ! serial_port.good() )
    {
        std::cerr << "Error: Could not set the character size." << std::endl ;
        exit(1) ;
    }
    //
    // Disable parity.
    //
    serial_port.SetParity( SerialStreamBuf::PARITY_NONE ) ;
    if ( ! serial_port.good() )
    {
        std::cerr << "Error: Could not disable the parity." << std::endl ;
        exit(1) ;
    }
    //
    // Set the number of stop bits.
    //
    serial_port.SetNumOfStopBits( 1 ) ;
    if ( ! serial_port.good() )
    {
        std::cerr << "Error: Could not set the number of stop bits."
                  << std::endl ;
        exit(1) ;
    }
    //
    // Turn off hardware flow control.
    //
    serial_port.SetFlowControl( SerialStreamBuf::FLOW_CONTROL_NONE ) ;
    if ( ! serial_port.good() )
    {
        std::cerr << "Error: Could not use hardware flow control."
                  << std::endl ;
        exit(1) ;
    }
    //
    // Do not skip whitespace characters while reading from the
    // serial port.
    //
    // serial_port.unsetf( std::ios_base::skipws ) ;
    //
    // Wait for some data to be available at the serial port.
    //
    while( serial_port.rdbuf()->in_avail() == 0 )
    {
        usleep(100) ;
    }
    //
    // Keep reading data from serial port and print it to the screen.
    //
    while( serial_port.rdbuf()->in_avail() > 0  )
    {
        char next_byte;
        serial_port.get(next_byte);
        std::cerr << std::hex << static_cast<int>(next_byte) << " ";
        usleep(100) ;
    }
    std::cerr << std::endl ;
    return EXIT_SUCCESS ;
}

crayzeewulf, 2016-01-26:
Can you please show the compiler command that you used to build your example program? Also, what is the output of the following command:

``ldd path_to_your_executable``

(Replace ``path_to_your_executable`` with the path to your program's executable, of course).

JohnBabrick, 2016-01-27:
I think I found my problem. 

I originally installed libserial from Ubuntu's repository which installed it to /usr/lib/x86_64-linux-gnu, then, when I couldn't get things to work correctly I tried removing it and then installing it by building the source from github. That installation put it in /usr/local/lib. However, apparently the remove did not remove it from /usr/lib/x86_64-linux-gnu and now I had versions in both /usr/lib/x86_64-linux-gnu (didn't work) and /usr/local/lib (does work). ldd showed a link to the /usr/lib/x86_64-linux-gnu.

The solution was to manually remove from /usr/lib and run ldconfig. For what it's worth I learned a bit about ldd, ldconfig and ld.so today. Thanks for your help, I look forward to using your code to access a ham radio's serial port.

crayzeewulf, 2016-01-27:
Excellent! Glad to hear that you found the source of the problem. Thanks for the update.

closed 2016-01-27 by crayzeewulf