crayzeewulf/libserial #59: Cross compile

bidasknakayama, 2016-02-27:
Hello All

Could you teach me How to do "cross" compile with "arm-linux-gnueabi-gcc, gcc++" ?
I want to know each step...

1) first is get tar.gz and extract.
2) configure ....-X-X-X-X options...
...

I need your knowledge.

crayzeewulf, 2016-02-28:
I do not have that cross compiler so I cannot test the cross-compilation process on my system. However, I can give you some hints to get you started. If you get in trouble, post the error message(s) and hopefully after a little back-and-forth we will have you going. 

* Clone the libserial Git repository using the command:
```
git clone https://github.com/crayzeewulf/libserial.git
```
* Prepare the source code for configuration:
```
cd libserial
make -f Makefile.dist
```
* Configure for cross compilation:
```
mkdir build-arm-linux-gnueabi
cd build-arm-linux-gnueabi
../configure --host arm-linux-gnueabi --prefix=/opt/arm-linux-gnueabi/
```
* Compile:
```
make
```
* Install:
```
make install
```

If all goes well, `libserial` will be installed under `/opt/arm-linux-gnueabi/`.

closed 2016-05-12 by crayzeewulf

iloveangpao, 2024-02-16:
Hi, I tried this method with aarch64-linux-gnu, but ran into the following problem:

```
make[2]: Entering directory '/home/libserial-1.0.0/build-aarch64-linux-gnu/test'
aarch64-linux-gnu-g++ -std=c++14 -DHAVE_CONFIG_H -I. -I../../test -I..  -I../../src  -Weffc++ -O2 -std=c++14 -MT unit_tests.o -MD -MP -MF .deps/unit_tests.Tpo -c -o unit_tests.o ../../test/unit_tests.cpp
mv -f .deps/unit_tests.Tpo .deps/unit_tests.Po
/bin/bash ../libtool  --tag=CXX   --mode=link aarch64-linux-gnu-g++ -std=c++14 -Weffc++ -O2 -std=c++14  -L/usr/local/lib -Wl,-rpath,/usr/local/lib -o unit_tests unit_tests.o ../src/libserial.la -lboost_unit_test_framework
libtool: link: aarch64-linux-gnu-g++ -std=c++14 -Weffc++ -O2 -std=c++14 -Wl,-rpath -Wl,/usr/local/lib -o .libs/unit_tests unit_tests.o  -L/usr/local/lib ../src/.libs/libserial.so -lboost_unit_test_framework -Wl,-rpath -Wl,/usr/lib/aarch64-linux-gnu/lib
aarch64-linux-gnu-g++ -std=c++14 -DHAVE_CONFIG_H -I. -I../../test -I..  -I../../src  -Weffc++ -O2 -std=c++14 -MT SerialPortUnitTests.o -MD -MP -MF .deps/SerialPortUnitTests.Tpo -c -o SerialPortUnitTests.o ../../test/SerialPortUnitTests.cpp
In file included from ../../test/SerialPortUnitTests.h:36,
                 from ../../test/SerialPortUnitTests.cpp:34:
../../test/UnitTests.h:40:10: fatal error: gtest/gtest.h: No such file or directory
   40 | #include <gtest/gtest.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:406: SerialPortUnitTests.o] Error 1
make[2]: Leaving directory '/home/libserial-1.0.0/build-aarch64-linux-gnu/test'
make[1]: *** [Makefile:467: all-recursive] Error 1
make[1]: Leaving directory '/home/libserial-1.0.0/build-aarch64-linux-gnu'
make: *** [Makefile:372: all] Error 2
```

As far as i know libserial compiles gtest together as an external project, and I am not supposed to install/compile gtest separately, did i do something wrong with my config?

```
../configure --host=aarch64-linux-gnu --prefix=/usr/lib/aarch64-linux-gnu/
```

```
export CROSS_COMPILE=aarch64-linux-gnu-
export CC=${CROSS_COMPILE}gcc
export CXX=${CROSS_COMPILE}g++
export LD=${CROSS_COMPILE}ld
export AR=${CROSS_COMPILE}ar
export AS=${CROSS_COMPILE}as
export RANLIB=${CROSS_COMPILE}ranlib
export CFLAGS="-O2"
export CXXFLAGS="-O2 -std=c++14"
export LDFLAGS="-L/usr/local/lib -Wl,-rpath,/usr/local/lib"
export CPLUS_INCLUDE_PATH=/usr/include/python3.8:/usr/local/include
```