crayzeewulf/libserial #62: Build failes on Arch linux systems

CountMurphy, 2016-04-05:
Running make fails with this error:
Making all in sip
make[2]: Entering directory '/tmp/libserial/sip'
python ./configure.py
  File "./configure.py", line 80
    """
      ^
SyntaxError: invalid syntax

I believe this is caused by configure.py needing python version 2 to run. If I run the command: 
python2 sip/configure.py I get the following output:

2016-04-05 16:08:44,362 INFO Build file is libserial.sbf
2016-04-05 16:08:44,363 INFO Reading sip configuration.
2016-04-05 16:08:44,363 INFO Running sip command: '/usr/bin/sip -c . -e -b libserial.sbf libserial.sip'
sip: Unable to find file "libserial.sip"
Error: Unable to open "/tmp/libserial/libserial.sbf"

On Arch systems, the command python defaults to version 3. If I run python2 sip/configure.py I get:
 File "sip/configure.py", line 80
    """
      ^
SyntaxError: invalid syntax

I have tried to change configure.py to use python2.7, but the make script still tries to execute with python 3.

crayzeewulf, 2016-04-07:
Thanks for posting the workaround!
-
CrayzeeWulf

lllars, 2017-08-27:
The work around works for me.  I needed to also install python2-sip though (pacman -S python2-sip).  Otherwise I get "ImportError: No module named sipconfig"

mcsauder, 2017-09-11:
Hi @CountMurphy and @lllars,

Would you be willing to try the latest [c++14 branch](https://github.com/crayzeewulf/libserial/tree/c%2B%2B14) to see if [this commit](https://github.com/crayzeewulf/libserial/pull/78/commits/1e487de0ad8ffa504c7ce70a1c19788d40fbdd1d) fixes the issue for you?

Let us know if it doesn't resolve the issue!

-Mark

CountMurphy, 2017-09-11:
The build fails with:

> LibSerialTests.cpp:23:10: fatal error: gtest/gtest.h: No such file or directory
>  #include <gtest/gtest.h>
>           ^~~~~~~~~~~~~~~
> compilation terminated.

No python errors though :)

mcsauder, 2017-09-11:
Hi @CountMurphy,

Do you have gtest installed or is is it possible for you to install gtest? (I'm not an arch linux user, so forgive me if it's not straightforward, I think you can use the package manager to install it.)  The current arch linux gtest version appears to be the same as Ubuntu [1.8](https://www.archlinux.org/packages/community/i686/gtest/).
`pacman -S gtest `

Let me know if that advances the build.

-Mark

CountMurphy, 2017-09-12:
Figured I was missing a dependency. The build goes further, but still fails.

> 
> make[2]: Leaving directory '/tmp/libserial/examples'
> Making all in test
> make[2]: Entering directory '/tmp/libserial/test'
> g++ -DHAVE_CONFIG_H -I. -I..  -I../src  -Weffc++ -g -O2 -MT unit_tests.o -MD -MP -MF .deps/unit_tests.Tpo -c -o unit_tests.o unit_tests.cpp
> mv -f .deps/unit_tests.Tpo .deps/unit_tests.Po
> /bin/sh ../libtool  --tag=CXX   --mode=link g++ -Weffc++ -g -O2   -o unit_tests unit_tests.o ../src/libserial.la -lboost_unit_test_framework 
> libtool: link: g++ -Weffc++ -g -O2 -o .libs/unit_tests unit_tests.o  ../src/.libs/libserial.so -lboost_unit_test_framework -Wl,-rpath -Wl,/usr/local/lib
> g++ -DHAVE_CONFIG_H -I. -I..  -I../src  -Weffc++ -g -O2 -MT LibSerialTests.o -MD -MP -MF .deps/LibSerialTests.Tpo -c -o LibSerialTests.o LibSerialTests.cpp
> LibSerialTests.cpp: In constructor ‘LibSerialTest::LibSerialTest()’:
> LibSerialTests.cpp:37:7: warning: ‘LibSerialTest::mutex’ should be initialized in the member initialization list [-Weffc++]
>  class LibSerialTest
>        ^~~~~~~~~~~~~
> LibSerialTests.cpp:37:7: warning: ‘LibSerialTest::serialStream1’ should be initialized in the member initialization list [-Weffc++]
> LibSerialTests.cpp:37:7: warning: ‘LibSerialTest::serialStream2’ should be initialized in the member initialization list [-Weffc++]
> LibSerialTests.cpp:37:7: warning: ‘LibSerialTest::serialPort1’ should be initialized in the member initialization list [-Weffc++]
> LibSerialTests.cpp:37:7: warning: ‘LibSerialTest::serialPort2’ should be initialized in the member initialization list [-Weffc++]
> In file included from /usr/include/gtest/gtest.h:58:0,
>                  from LibSerialTests.cpp:23:
> LibSerialTests.cpp: In constructor ‘LibSerialTest_testSerialStreamOpenClose_Test::LibSerialTest_testSerialStreamOpenClose_Test()’:
> LibSerialTests.cpp:1557:1: note: synthesized method ‘LibSerialTest::LibSerialTest()’ first required here 
>  TEST_F(LibSerialTest, testSerialStreamOpenClose)
>  ^
> mv -f .deps/LibSerialTests.Tpo .deps/LibSerialTests.Po
> make[2]: *** No rule to make target '/usr/lib/libgtest.a', needed by 'LibSerialTests'.  Stop.
> make[2]: Leaving directory '/tmp/libserial/test'
> make[1]: *** [Makefile:465: all-recursive] Error 1
> make[1]: Leaving directory '/tmp/libserial'
> make: *** [Makefile:370: all] Error 2
> 

If its any consolation, all of the .cpp files in src have corresponding .o files so I'm pretty sure it builds (just not links).

mcsauder, 2017-09-12:
Hi @CountMurphy,

Thanks for working through this with me.  I am curious if you try again if it might succeed simply as a result of multiple cores/multiple make threads being invoked and not everything getting built in the right order.  Also, next, would you try using cmake to build it, (provided you have cmake installed).  There is a compile script you can simply run, `./compile.sh`.

Let me know the output from those attempts!  Thanks again for trying this.

-Mark

mcsauder, 2017-09-12:
Ah, also, it could be that there is a variant on gtest that needs to be installed, something like gtest-dev or libgtest-dev, (Debian packages require libgtest-dev).

CountMurphy, 2017-09-12:
Builds wonderfully with cmake (and having gtest installed). Though I'm not using bash by default and had to add a #!/bin/bash to the compile.sh script (actually using bash I didn't need to add it).

mcsauder, 2017-09-12:
Cool, that's great news!  I'll talk to Crayzee Wulf and see if he has any thoughts about what might be the hang up with make.  The top level CMakeLists.txt is pulling down GTest and building it, which is why it is likely succeeding where make didn't.  I'll do some more testing to try to understand what's going on with make.  Thanks for your feedback!

-Mark

CountMurphy, 2017-09-12:
No problem, glad to help!

mcsauder, 2017-09-12:
Hi Everyone, 
As a follow-up, here is a link describing the reason make didn't work initially: [https://askubuntu.com/questions/145887/why-no-library-files-installed-for-google-test](https://askubuntu.com/questions/145887/why-no-library-files-installed-for-google-test)

crayzeewulf, 2017-09-12:
Very interesting. Thanks for digging this up!

crayzeewulf, 2017-09-13:
If you guys are happy with the status, is it okay to close this issue at this point?

mcsauder, 2017-09-13:
Hi @lllars,

In case it simplifies things for you to test the fix, you can replace the print statement on [this line](https://github.com/crayzeewulf/libserial/blob/fbb770f5ddfec3e13cba74a30bff80803c15e56c/sip/configure.py.in#L78) with the print() method call on [this line](https://github.com/crayzeewulf/libserial/blob/39bdf19806e8448c0f0baef7a29500e261af6826/sip/configure.py.in#L77) in the sip/configure.py.in file of the master branch.  This will allow you to build as you are normally used to.  Alternatively, checkout the c++14 branch and build with cmake by using the compile script.

Let us know if this fixes things for you!

-Mark

mcsauder, 2017-09-23:
Fixed in the master branch with PR #82!  Let us know if for any reason the issue persists.

-Mark

closed 2017-09-23 by mcsauder