crayzeewulf/libserial #80: CMake Include Target property

Klirxel, 2017-09-22:
Hello libserialTeam

I cant integrate libserial in my CMake project without workaround.
Include files are not added to the public interface for LibSerial.

Please add to src/CMakeLists.txt

`target_include_directories (LibSerial PUBLIC ${CMAKE_CURRENT_LIST_DIR})
`

mcsauder, 2017-09-22:
Hi Klirxel,

Thanks for posting the issue.  I've implemented your solution and pushed the commit into the (currently active) PR #79 .

Thanks for the patch hint!

-Mark

mcsauder, 2017-09-23:
Hi Klirxel,

I think everything is fixed up now in both the development branch and the master branch so I'll close this issue.  Please let us know if you spot anything else that can be improved!

-Mark

closed 2017-09-23 by mcsauder