crayzeewulf/libserial #89: Build issues

stephamd, 2017-10-28:
when building this library in my ubuntu vm I receive the following build errors:

SerialPort.cpp: In member function ‘void SerialPort::SerialPortImpl::Open()’:
SerialPort.cpp:647:48: error: ‘errno’ was not declared in this scope
         throw SerialPort::OpenFailed( strerror(errno) )  ;
                                                ^
SerialPort.cpp:663:48: error: ‘errno’ was not declared in this scope
         throw SerialPort::OpenFailed( strerror(errno) ) ;
                                                ^
SerialPort.cpp:673:48: error: ‘errno’ was not declared in this scope
         throw SerialPort::OpenFailed( strerror(errno) ) ;
                                                ^
SerialPort.cpp:683:48: error: ‘errno’ was not declared in this scope
         throw SerialPort::OpenFailed( strerror(errno) ) ;
                                                ^
SerialPort.cpp:715:48: error: ‘errno’ was not declared in this scope
         throw SerialPort::OpenFailed( strerror(errno) ) ;
                                                ^
etc...

I installed the libgtest-dev libboost-dev libs but errno would not have anything to do with that.  Not sure if this is a bug or if I am missing something obvious.  I have experience building libraries but have not ran into this issue before.

Thanks

mcsauder, 2017-10-29:
Hi Matt,

Thanks for entering the issue.  Would you mind adding 
`#include <errno.h>`

to SerialPort.cpp and let me know if that fixes it for you?  I'm not sure why the VM doesn't have that file by default, but if it fixes the issue for you I'll add it explicitly to the current C++14 branch pull request and just push a commit up to the master branch as well.

Let me know once you've added that line if it fixes things up!

-Mark

mcsauder, 2017-10-31:
Hi @stephamd ,

I've merged PR #90 into the master branch, and added a commit to the current c++14 branch PR #88 to the same effect.  Could you let us know if adding std:: to the strerror calls fixes your issue?  Also, would you let us know if the issue persists with the current code and if adding _#include <cerrno>_ fixes it?  Thanks!

-Mark

stephamd, 2017-10-31:
I added:
SerialPort.cpp:
#include <errno.h>
#include <stdlib.h>

PosixSignalDispatcher.cpp:
#include <errno.h>

It was able to build, however I got errors when it reached the libserial/test directory.  I will try some things and let you know what may be happening.  What system are you developing on?

Thanks

mcsauder, 2017-10-31:
Hi @stephamd ,

I am developing on Ubuntu 17.04 and 17.10 using GCC6.3.0 and GCC7.2.0.

Would you post the errors you see in the test directory?  Thanks!

-Mark

mcsauder, 2017-11-03:
Hi @stephamd ,

Are you using the make files or the compile script/CMake to build? Can post the build errors you are seeing in the test directory?  I can look into things more once you post the build errors you are observing.

Thanks!

BlockByBlock, 2017-11-06:
Hi, I encountered the same errors. However, upon adding 
#include <errno.h>
#include <stdlib.h>
The problem is fixed. I am on GCC 5.4.0

mcsauder, 2017-11-06:
Hi @BlockByBlock ,

Thanks for adding to the thread, I appreciate it.  

Since it fixes the issue I'll go ahead and add in the additional #include statements to the master and c++14 branches.  

Because the root of the issue still evades me, would you post the errors you see once you've added only #include <cerrno> (without adding #include <stdlib.h>)?  I can't spot why this additional include is required without being able to read the compiler errors, and I am curious if the reason I don't see these error is because my machine has had libc6-dev installed previously.

Last, would you be willing to see if installing libc6-dev fixes those errors without adding #include <stdlib.h>? 
`sudo apt install libc6-dev`

Thanks for your help!

-Mark

BlockByBlock, 2017-11-07:
Hi Mark,

The error for without stdlib is: 
SerialPort.cpp:1730:34: error: 'abs' was not declared in this scope
         if ( abs( result.tv_usec ) > MICROSECONDS_PER_SECOND )

I do have 'build-essential' and 'libc6-dev' installed before the make and still encounter the same error.

I am testing on Ubuntu 16.04 docker image, with GCC 5.4.0. Will test on other platform later. I am thinking to up my GCC to 6.3.0 and see if the problem persists, keep you updated again.

mcsauder, 2017-11-07:
Thanks @BlockByBlock !

I've merged the fix into the master branch with PR #91 and the c++14 branch with PR #92, (although with the changes in the c++14 branch already in place that issue likely wouldn't have been encountered).  Thanks for your work investigating this, it's fascinating that it arose in the manner it has.

As a side note, I've seen a set of compiler warnings show up with GCC 7.2.0 that don't arise with earlier GCC warnings.  Work underway is to push out a new release candidate from the c++14 branch. The next release fixes the multi-threading issue, adds set/get methods to the SerialStream class that exist in the SerialPort class, and standardizes method naming between SerialPort and SerialStream classes, (along with a lot of documentation).

Let us know if the current master branch and/or c++14 branch fixes things up for you, or if you learn anything else along the way.

Thanks again!

-Mark

mcsauder, 2017-11-11:
Hi @stephamd and @BlockByBlock ,

Do the PR's #91, and #92 fix the issue for you?  Are we good to close this issue?

Thanks again for reporting the bug and figuring out the fixes for it!

-Mark

BlockByBlock, 2017-11-12:
@mcsauder yup, all good, you can this issue

mcsauder, 2017-11-12:
Thanks again! Closed!

-Mark

closed 2017-11-12 by mcsauder