crayzeewulf/libserial #93: SerialPort::Read() does not support timeout

wade-fs, 2017-12-01:
--- /home/wade/src/SerialPort/libserial/src/SerialPort.cpp      2017-12-01 11:58:19.114891152 +0800
+++ SerialPort.cpp      2017-11-30 17:39:41.865658561 +0800
@@ -33,8 +33,12 @@
 #include <sys/ioctl.h>
 #include <sys/time.h>
 #include <unistd.h>
+#include <asm/socket.h>

-namespace
+using namespace LibSerial ;
+using namespace std ;
+
+namespace LibSerial
 {
     //
     // Various error messages used in this file while throwing
@@ -319,6 +323,12 @@
                std::runtime_error ) ;
 } ;

+SerialPort::SerialPort() :
+    mSerialPortImpl(new SerialPortImpl("/dev/ttyS0") )
+{
+    /* empty */
+}
+
 SerialPort::SerialPort( const std::string& serialPortName ) :
     mSerialPortImpl(new SerialPortImpl(serialPortName) )
 {
@@ -1454,10 +1464,9 @@
         //
         // Read all available data if numOfBytes is zero.
         //
-        while( this->IsDataAvailable() )
-        {
+        do {
             dataBuffer.push_back( ReadByte(msTimeout) ) ;
-        }
+        } while( this->IsDataAvailable() );
     }
     else
     {
@@ -1703,7 +1712,7 @@
     return ;
 }

-namespace
+namespace LibSerial
 {
     const struct timeval
     operator-( const struct timeval& firstOperand,

mcsauder, 2017-12-01:
Hi Wade,

Thank you for submitting your issue.

I'm having a little bit of difficulty understanding what part might be concerning to you.  Here are the function prototypes for all SerialPort class Read() related methods:
```

        void Read(char&        charBuffer,
                  const size_t numberOfBytes = 0,
                  const size_t msTimeout  = 0);

        void Read(unsigned char& charBuffer,
                  const size_t   numberOfBytes = 0,
                  const size_t   msTimeout  = 0);

        void Read(DataBuffer&  dataBuffer,
                  const size_t numberOfBytes = 0,
                  const size_t msTimeout  = 0);

        void Read(std::string& dataString,
                  const size_t numberOfBytes = 0,
                  const size_t msTimeout  = 0);

        void ReadByte(char&        charBuffer,
                      const size_t msTimeout = 0);

        void ReadByte(unsigned char& charBuffer,
                      const size_t   msTimeout = 0);

        void ReadLine(std::string&  dataString,
                      const char    lineTerminator = '\n',
                      const size_t  msTimeout = 0);
```

As you can see, each one has a timeout value that you can specify of size_t type, (unsigned int), so that you can tell it how long the timeout value should be.

Is the issue you see that a ReadTimeout exception error is being thrown?  If so, that is the design.  If you do not receive data within your timeout period it will throw an exception which you can "catch".

There are unit tests verifying the functionality beginning here, [https://github.com/crayzeewulf/libserial/blob/12012111ce8b89375328c5307d100ce7091eb2cb/test/UnitTests.cpp#L840](url), so perhaps if you could help us a little more with more description of your problem we might be able to help.

Please help me better understand the issue you are seeing, and thank you for letting us know about your issue!

-Mark

crayzeewulf, 2017-12-01:
Thanks, Mark. 

I edited the original comment by @wade-fs a little bit so it is formatted correctly in Github Markdown but the contents are essentially unchanged. I am looking forward to receiving additional information about this issue from @wade-fs .

wade-fs, 2017-12-01:
This is "implementation issue", not declare issue.

Please change while() { ..} to do { ... } while();
because in original, the "while( this->IsDataAvailable() )" 
it will be failed bcz there is no any data available(), but not "timeouted"

mcsauder, 2017-12-01:
Hi @wade-fs ,

Could you help me localize this issue between the master branch and the c++14 branch? Could you tell me if the problem exists in both the master and the c++14 branches? I'm not quite understanding how changing the while() loop to a do(){...} while() loop is fixing the issue.

Thank you for helping me localize and understand what you are seeing.

-Mark

ADDITIONAL: @wade-fs ,  (noted in PR #95)
There is a known issue with the master branch in threaded applications. The master branch will not function as expected in a threaded application. If you are using the master branch in a threaded application could you try the c++14 branch and let us know if that resolves the issue? Thank you for your work on this!

-Mark

crayzeewulf, 2017-12-01:
@wade-fs I believe that the implemented behavior is intentional and matches the documentation. As the documentation says:

> If numberOfBytes is zero, then this method will keep reading data till no more data is available at the serial port. 

The intention is to return all *currently available* data when `numberOfBytes` is set to 0. If no data is  available then it should return immediately and `dataBuffer` should be empty. 

Hope this makes sense. If not, please give us an example of how you are intending to use this method and what you expect it to do. Maybe there is another way to achieve the affect you want.

wade-fs, 2017-12-01:
the example:

while (serial_port.Read(buf)) {
  cout << buf << endl;
}
In your case, this will meaningless....

mcsauder, 2017-12-01:
Hi @wade-fs ,

I think you have discovered a very fun corner case, but I think to do it service it might require a bit more time and thought...

Let me see if I can describe what you are thinking about... I think you are describing a use case where an unknown amount of data might be arriving, but once data begins arriving you would like to be able to apply a timeout value to determine when data has stopped.  The special case here is that you would like the call to block until data arrives and then be able to apply the timeout... Does that seem to describe what you are hoping to see happen?

If this is the case, perhaps  youmight actually be best served by utilizing the SerialStream class, which is a blocking class by default...  this class might get you what you desire without forcing the SerialPort Class to block until some data has arrived.

In the c++14 branch there is a private method in the SerialPort::Implementation class that is also capable of achieving that goal, however, it is private to the implementation class.

Give us a bit of time to try to best accommodate this if I am interpreting everything correctly, and if I am not please let me know!

-Mark

crayzeewulf, 2017-12-01:
> while (serial_port.Read(buf)) {
>     cout << buf << endl;
> }

@wade-fs `serial_port.Read(buf)` does not return anything (`void`). So the above loop will not compile. It will be helpful if you could provide an example of how you are intending to use this method. In the mean time, please see my comments below based on how @mcsauder described your probable use case.  

The above Read() method will return data only if some data is available at the serial port at the moment it is called. Serial port hardware and/or the kernel driver for that hardware will typically have a small buffer/fifo that holds incoming data. If there is anything waiting in that buffer (see `FIONREAD` [here](http://www.cmrr.umn.edu/~strupp/serial.html#5_1_3)) then this method will keep reading data as long as that buffer is not empty. However, if there is no data in this buffer when the method is called, the intention was to return immediately with an empty buffer indicating that there is no data ready to be read. This was the original intention and was driven by a specific use case when this library was first written (a while ago). But this may not be what makes sense in your application. 

Having said that and based on @mcsauder's comment, it appears like you are trying to do the following (please correct or explain if this is not true) by setting `numberOfBytes` to 0 and `msTimeout` to a non-zero value:

1. If data is available, start reading it and keep reading it as long as more keeps coming. If no new data arrives for `msTimeout` ms then give up and return everything that was read so far. 
1.  If no data is available, wait as much as `msTimeout` ms for it to arrive. If no data arrives in this much time then give up and return an empty buffer. Otherwise, if data does arrive in `msTimeout` ms, do as in the step#1 above. 

_In the current implementation_, there is no simple way to achieve this that I can think of. You can only achieve something like this indirectly. For example:

```cpp
while(not serial_port.IsDataAvailable())
{
    // do something else useful or sleep for a bit. Break out if data is not available in msTimeout ms. 
}
serial_port.Read(buf, 0, msTimeout) ;
```

On the other hand, switching `while() {...}` with `do { ... } while(..)` as you suggested will achieve this behavior directly from the `Read()` method. I do not see any issue with this modification and I do not know of any code that depends on the original behavior any more. If the above is what you intended and accurately represents your use case, please let us know.

wade-fs, 2017-12-01:
>    2. If no data is available, wait as much as msTimeout ms for it to

Ok, I want to read any data, timeout is 10 sec:

case 1:
      <---- remote send data
then Read()
==> return immediately

case 2:
  Read() first,
then <---- remote send data
==> return immediately too

I cannot use Read() to read data .

Best Regards
--
自由的精靈, 狂想的空間
Free Spirit, Fantasy Space

mcsauder, 2017-12-01:
Hi @wade-fs ,

I think I finally understand your use case.  Please let me know if PR #97 matches the behavior (and documentation in the header file) you are expecting.

Thanks for your work on this!

-Mark

mcsauder, 2017-12-19:
Hi @wade-fs,

I just wanted to thank you for posting your issue.  We have done additional [work](https://github.com/crayzeewulf/libserial/pull/97) in the code that should incorporate your feedback as well as help prevent memory overflows as well as speed up the code as well.  This work wouldn't have been undertaken without your feedback, so, thanks once again!

-Mark

mcsauder, 2018-01-03:
Hi @wade-fs ,

PR #97 has been accepted into the c++14 branch and fixes the Read() behavior to give the result you described.  The decision was made to deprecate the Read(char*) and Read(unsigned char*) methods to eliminate the opportunity for array out of bounds runtime errors.  The ReadyByte(char) Read(std::string), and Read(DataBuffer) methods have all been updated with the fix for this issue and refactored to eliminate the difference in speed previously observed in the methods.  The Read() methods all perform faster than previous and at the same speed the Read(char*) methods achieved.

Thanks again for reporting the issue!  Let us know if this takes care of everything and we can close it out.

-Mark

closed 2018-01-09 by mcsauder