crayzeewulf/libserial #98: make does not work on raspbian

ktphys, 2017-12-03:
I copied the actual version of libserial onto my RPI3 with the actual Raspian version ( Debian GNU/Linux) and gcc etc. all updated.

"configure" works, but not the generated "Makefile".
Here is the output, which marks some problems with the option "CXX":
________

$ make
make  all-recursive
make[1]: Entering directory '/usr/src/libserial-0.6.0rc2'
Making all in src
make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/src'
/bin/sh ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c -o SerialStreamBuf.lo SerialStreamBuf.cc
../libtool: 1578: ../libtool: preserve_args+= --tag CXX: not found
../libtool: 1: eval: base_compile+= g++: not found
../libtool: 1: eval: base_compile+= -DHAVE_CONFIG_H: not found
../libtool: 1: eval: base_compile+= -I.: not found
../libtool: 1: eval: base_compile+= -I..: not found
../libtool: 1: eval: base_compile+= -Weffc++: not found
../libtool: 1: eval: base_compile+= -g: not found
../libtool: 1: eval: base_compile+= -O2: not found
../libtool: 1: eval: base_compile+= -MT: not found
../libtool: 1: eval: base_compile+= SerialStreamBuf.lo: not found
../libtool: 1: eval: base_compile+= -MD: not found
../libtool: 1: eval: base_compile+= -MP: not found
../libtool: 1: eval: base_compile+= -MF: not found
../libtool: 1: eval: base_compile+= .deps/SerialStreamBuf.Tpo: not found
../libtool: 1: eval: base_compile+= -c: not found
libtool: compile: you must specify a compilation command
libtool: compile: Try `libtool --help --mode=compile' for more information.
Makefile:445: recipe for target 'SerialStreamBuf.lo' failed
make[2]: *** [SerialStreamBuf.lo] Error 1
make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/src'
Makefile:453: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/usr/src/libserial-0.6.0rc2'
Makefile:357: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $
___________
Sorry, I'm not familiar enough with gcc on Linux to resolve this problem - could please anybody give me some hint?
Thank you for your help!

mcsauder, 2017-12-03:
Hi @ktphys ,

Can you see which version of GCC you have installed?
`gcc -v`

If there isn't a version of GCC already installed, try the installing build-essential and g++ and let us know if it fixes anything for you!
`sudo apt-get install build-essential g++`

-Mark

ktphys, 2017-12-03:
Hello Mark,

 

it‘s gcc version 4.9.2  (Raspbian 4.9.2-10).

 

Here is the full output of gcc –v:

$ gcc -v

Using built-in specs.

COLLECT_GCC=gcc

COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper

Target: arm-linux-gnueabihf

Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf

Thread model: posix

gcc version 4.9.2 (Raspbian 4.9.2-10)

 

 

Best regards

Klaus

crayzeewulf, 2017-12-03:
> ../libtool: 1: eval: base_compile+= g++: not found

@ktphys It looks like `g++` is probably not installed on your system. Please try out the `apt-get` command that @mcsauder  mentioned:

```sh
sudo apt-get install build-essential g++
```

After that you ought to be able to run the following command to check if `g++` is installed and which version you have:

```sh
g++ --version
```

After `g++` is installed, please try the build again and let us know how that goes.

mcsauder, 2017-12-03:
Hi @ktphys ,

As crayzeeWulf mentioned, let us know if installing build-essential and g++ resolves your issue.  If it doesn't, comb through the terminal output carefully when you run the first two build commands, and let us know if you see any errors:

`make -F Makefist.dist
./configure`

Next, can you query your libtool version and location and let us know what they are:

`libtool -version
which libtool`

My machine's responses to those queries are:
  libtool (GNU libtool) 2.4.6
  /usr/bin/libtool

If your libtool queries are unsuccessful please install libtool:

`sudo apt install libtool-bin`

Once you've done all of that let us know if it's still not working and we'll keep digging into it with you.

Thanks for working on it with us!

-Mark

ktphys, 2017-12-03:
Hello,

 

in my opinion, g++ is installed. 

I tried 

sudo apt-get install build-essential g++

and get the following:

 

sudo apt-get install build-essential g++

Paketlisten werden gelesen... Fertig

Abhängigkeitsbaum wird aufgebaut.

Statusinformationen werden eingelesen.... Fertig

build-essential ist schon die neueste Version.

g++ ist schon die neueste Version.

g++ wurde als manuell installiert festgelegt.

Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:

  libasn1-8-heimdal libcwiid1 libgssapi3-heimdal libhcrypto4-heimdal

  libheimbase1-heimdal libheimntlm0-heimdal libhx509-5-heimdal libjs-prettify

  libkrb5-26-heimdal libllvm3.7 libqscintilla2-11 libqscintilla2-l10n

  libqt4-network libqt4-xmlpatterns libqtwebkit4 libroken18-heimdal

  libruby1.9.1 libruby1.9.1-dbg libtcl8.5 libtcltk-ruby1.9.1 libtk8.5

  libwind0-heimdal libxfce4ui-1-0 pypy-upstream-doc ri1.9.1 ruby1.9.1

  ruby1.9.1-dev ruby1.9.1-examples ruby1.9.1-full ruby1.9.3 supercollider

  supercollider-common supercollider-ide supercollider-language

  supercollider-supernova tcl8.5 tk8.5 wiringpi xfce-keyboard-shortcuts

Verwenden Sie »apt-get autoremove«, um sie zu entfernen.

0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

 

 

g++ --version yields:

 

g++ (Raspbian 4.9.2-10) 4.9.2

Copyright (C) 2014 Free Software Foundation, Inc.

This is free software; see the source for copying conditions.  There is NO

warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

 

 

Klaus

ktphys, 2017-12-03:
Hello Mark,

 

make -f Makefile.dist ./configure  yields:

 

make: 'configure' is up to date.

 

Libtool is definitely not built and installed:

 

libtool -version which libtool     yields:

-bash: libtool: command not found

 

Any suggestions?

 

Klaus

mcsauder, 2017-12-03:
Hi @ktphys ,

Thanks for the information.  Please install libtool:

`sudo apt install libtool-bin`

Let us know if it installs correctly and if that fixes your issue!

-Mark

ktphys, 2017-12-03:
Hi Mark,

 

I applied again 

 

sudo apt install libtool-bin

 

Paketlisten werden gelesen... Fertig

Abhängigkeitsbaum wird aufgebaut.

Statusinformationen werden eingelesen.... Fertig

libtool-bin ist schon die neueste Version.

Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:

  libasn1-8-heimdal libcwiid1 libgssapi3-heimdal libhcrypto4-heimdal

  libheimbase1-heimdal libheimntlm0-heimdal libhx509-5-heimdal libjs-prettify

  libkrb5-26-heimdal libllvm3.7 libqscintilla2-11 libqscintilla2-l10n

  libqt4-network libqt4-xmlpatterns libqtwebkit4 libroken18-heimdal

  libruby1.9.1 libruby1.9.1-dbg libtcl8.5 libtcltk-ruby1.9.1 libtk8.5

  libwind0-heimdal libxfce4ui-1-0 pypy-upstream-doc ri1.9.1 ruby1.9.1

  ruby1.9.1-dev ruby1.9.1-examples ruby1.9.1-full ruby1.9.3 supercollider

  supercollider-common supercollider-ide supercollider-language

  supercollider-supernova tcl8.5 tk8.5 wiringpi xfce-keyboard-shortcuts

Verwenden Sie »apt-get autoremove«, um sie zu entfernen.

0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

 

==> Everything was already installed, nothing was renewed

 

“make” still does not work:

 

make

make  all-recursive

make[1]: Entering directory '/usr/src/libserial-0.6.0rc2'

Making all in src

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/src'

/bin/sh ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c -o SerialStreamBuf.lo SerialStreamBuf.cc

../libtool: 1578: ../libtool: preserve_args+= --tag CXX: not found

../libtool: 1: eval: base_compile+= g++: not found

../libtool: 1: eval: base_compile+= -DHAVE_CONFIG_H: not found

../libtool: 1: eval: base_compile+= -I.: not found

../libtool: 1: eval: base_compile+= -I..: not found

../libtool: 1: eval: base_compile+= -Weffc++: not found

../libtool: 1: eval: base_compile+= -g: not found

../libtool: 1: eval: base_compile+= -O2: not found

../libtool: 1: eval: base_compile+= -MT: not found

../libtool: 1: eval: base_compile+= SerialStreamBuf.lo: not found

../libtool: 1: eval: base_compile+= -MD: not found

../libtool: 1: eval: base_compile+= -MP: not found

../libtool: 1: eval: base_compile+= -MF: not found

../libtool: 1: eval: base_compile+= .deps/SerialStreamBuf.Tpo: not found

../libtool: 1: eval: base_compile+= -c: not found

libtool: compile: you must specify a compilation command

libtool: compile: Try `libtool --help --mode=compile' for more information.

Makefile:445: recipe for target 'SerialStreamBuf.lo' failed

make[2]: *** [SerialStreamBuf.lo] Error 1

make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/src'

Makefile:453: recipe for target 'all-recursive' failed

make[1]: *** [all-recursive] Error 1

make[1]: Leaving directory '/usr/src/libserial-0.6.0rc2'

Makefile:357: recipe for target 'all' failed

make: *** [all] Error 2

 

Best regards

 

Klaus

mcsauder, 2017-12-03:
Hi @ktphys ,

if the libtool queries "libtool -version" and "which libtool" don't return any values, it would seem that your system path might not be recognizing your libtool installation.  

Let me know if I am misunderstood and your system is recognizing libtool.

I have found a stack exchange post on that topic here: [https://askubuntu.com/questions/625523/libtool-installed-but-not-on-path-after-installation](https://askubuntu.com/questions/625523/libtool-installed-but-not-on-path-after-installation)

As suggested in the post, it might also be worth running an apt update and upgrade followed by a reboot, as suggest by a response in that same post.

`sudo apt update`
`sudo apt upgrade`
`sudo reboot`

If your machine is not recognizing libtool, try following the suggestions in the post and let us know if anything there is successful at getting the system to recognize libtool.

While on the topic, I was curious if you might have tried compiling with CMake using the _compile.sh_ script?

Thanks for troubleshooting this with us and reporting your results.

-Mark

ktphys, 2017-12-03:
Hello Mark,

 

I did again the update, upgrade and reboot without any change: “make” is not working:

 

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $ make

make  all-recursive

make[1]: Entering directory '/usr/src/libserial-0.6.0rc2'

Making all in src

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/src'

/bin/sh ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c -o SerialStreamBuf.lo SerialStreamBuf.cc

../libtool: 1578: ../libtool: preserve_args+= --tag CXX: not found

../libtool: 1: eval: base_compile+= g++: not found

../libtool: 1: eval: base_compile+= -DHAVE_CONFIG_H: not found

../libtool: 1: eval: base_compile+= -I.: not found

../libtool: 1: eval: base_compile+= -I..: not found

../libtool: 1: eval: base_compile+= -Weffc++: not found

../libtool: 1: eval: base_compile+= -g: not found

../libtool: 1: eval: base_compile+= -O2: not found

../libtool: 1: eval: base_compile+= -MT: not found

../libtool: 1: eval: base_compile+= SerialStreamBuf.lo: not found

../libtool: 1: eval: base_compile+= -MD: not found

../libtool: 1: eval: base_compile+= -MP: not found

../libtool: 1: eval: base_compile+= -MF: not found

../libtool: 1: eval: base_compile+= .deps/SerialStreamBuf.Tpo: not found

../libtool: 1: eval: base_compile+= -c: not found

libtool: compile: you must specify a compilation command

libtool: compile: Try `libtool --help --mode=compile' for more information.

Makefile:445: recipe for target 'SerialStreamBuf.lo' failed

make[2]: *** [SerialStreamBuf.lo] Error 1

make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/src'

Makefile:453: recipe for target 'all-recursive' failed

make[1]: *** [all-recursive] Error 1

make[1]: Leaving directory '/usr/src/libserial-0.6.0rc2'

Makefile:357: recipe for target 'all' failed

make: *** [all] Error 2

 

In my opinion, it is not a question of incomplete path – it must be something with the “--tag CXX: not found” problem.

 

Best regards

Klaus

mcsauder, 2017-12-03:
Hi @ktphys ,

Sorry to ask once again, but can you copy/paste the terminal output to the following commands back in this post for me?

`libtool -version`
`which libtool`

and also:

`whereis libtool`
`locate libtool`

Thanks.

-Mark

ktphys, 2017-12-04:
Hello Mark,

 

here is the response to the commands you asked me yesterday to issue:

 

pi@raspberryKT:~ $ which libtool

/usr/bin/libtool

pi@raspberryKT:~ $ whereis libtool

libtool: /usr/bin/libtool /usr/share/libtool /usr/share/man/man1/libtool.1.gz /usr/src/libserial-0.6.0rc2/libtool

pi@raspberryKT:~ $ locate libtool

-bash: locate: Kommando nicht gefunden.

pi@raspberryKT:~ $

 

pi@raspberryKT:~ $ libtool --version

libtool (GNU libtool) 2.4.2

Written by Gordon Matzigkeit <gord@gnu.ai.mit.edu>, 1996

 

Copyright (C) 2011 Free Software Foundation, Inc.

This is free software; see the source for copying conditions.  There is NO

warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

 

Best regards

Klaus

mcsauder, 2017-12-04:
Hi @ktphys ,

Can you tell me what shell your machine is using for default:

`echo $SHELL`

my machine gives: /bin/bash

ktphys, 2017-12-04:
The same on my raspi: /bin/bash

 

Greetings

Klaus

mcsauder, 2017-12-04:
Hi @ktphys ,

Here are a few links that all seem to point to the issue being the use of a shell less sophisticated than bash, (even though your default shell appears to be bash), and this is the direction I am focusing right now:

- [https://lists.gnu.org/archive/html/bug-libtool/2011-09/msg00005.html](https://lists.gnu.org/archive/html/bug-libtool/2011-09/msg00005.html)

- [https://sourceforge.net/p/libint/tickets/14/](https://sourceforge.net/p/libint/tickets/14/)

- [https://github.com/fangism/hackt/issues/21](https://github.com/fangism/hackt/issues/21)

If you open up your user profile you might be able to see if your system is utilizing a shell binary from your home directory.

`gedit ~/.profile`

It would appear to me that the shell mis-match is the likely culprit.

I am a little uncertain if this is correct, but please try adding this line to your configure.ac file:

`SHELL = /bin/sh`

Let me know that changes anything.

-Mark

ktphys, 2017-12-04:
Hello,

I switched temporarily to /bin/sh and started make – no difference, I get the same errors:

 

$ make

make  all-recursive

make[1]: Entering directory '/usr/src/libserial-0.6.0rc2'

Making all in src

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/src'

/bin/sh ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c -o SerialStreamBuf.lo SerialStreamBuf.cc

../libtool: 1578: ../libtool: preserve_args+= --tag CXX: not found

../libtool: 1: eval: base_compile+= g++: not found

../libtool: 1: eval: base_compile+= -DHAVE_CONFIG_H: not found

../libtool: 1: eval: base_compile+= -I.: not found

../libtool: 1: eval: base_compile+= -I..: not found

../libtool: 1: eval: base_compile+= -Weffc++: not found

../libtool: 1: eval: base_compile+= -g: not found

../libtool: 1: eval: base_compile+= -O2: not found

../libtool: 1: eval: base_compile+= -MT: not found

../libtool: 1: eval: base_compile+= SerialStreamBuf.lo: not found

../libtool: 1: eval: base_compile+= -MD: not found

../libtool: 1: eval: base_compile+= -MP: not found

../libtool: 1: eval: base_compile+= -MF: not found

../libtool: 1: eval: base_compile+= .deps/SerialStreamBuf.Tpo: not found

../libtool: 1: eval: base_compile+= -c: not found

libtool: compile: you must specify a compilation command

libtool: compile: Try `libtool --help --mode=compile' for more information.

Makefile:445: recipe for target 'SerialStreamBuf.lo' failed

make[2]: *** [SerialStreamBuf.lo] Error 1

make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/src'

Makefile:453: recipe for target 'all-recursive' failed

make[1]: *** [all-recursive] Error 1

make[1]: Leaving directory '/usr/src/libserial-0.6.0rc2'

Makefile:357: recipe for target 'all' failed

make: *** [all] Error 2

$

 

 

Do you think it makes a difference if I alter the default shell in my profile?

 

Klaus

 

*         https://lists.gnu.org/archive/html/bug-libtool/2011-09/msg00005.html

*         https://sourceforge.net/p/libint/tickets/14/

*         fangism/hackt#21 <https://github.com/fangism/hackt/issues/21>

mcsauder, 2017-12-04:
Hi @ktphys ,

Apologies, I made a type in my first post and immediately corrected it, but it appears I was too late.

Please use the bash shell: SHELL = /bin/bash

I think that bin/sh is the issue, we need it to use bin/bash.  Both the Bourne shell, sh, or dash, which Raspbian uses as well, are both less sophisticated than bash, (Bourne Again SHelll).  Neither sh or dash recognize the compund "+=" statements, which is why libtool needs bash.

Work on getting libtool to recognize your bash installation and I think you'll be able to get past this error.

-Mark

mcsauder, 2017-12-04:
Hi again @ktphys ,

There is an outside chance that upgrading your libtool to a newer version could also help this issue.  If you are not averse to doing a dist-upgrade to your packages, this might also have an impact.

`sudo apt -y update && sudo apt -y dist-upgrade'

ktphys, 2017-12-04:
Hello,

I issued the command suggested by you (`sudo apt -y update && sudo apt -y dist-upgrade'), but the system found nothing to update – everything is on the latest release.

Also, my standard shell is “bash” – so I did not understand your lase e-mail.

Best regards

Klaus

mcsauder, 2017-12-04:
Hi @ktphys ,

Please read through the links I posted:

* [https://lists.gnu.org/archive/html/bug-libtool/2011-09/msg00005.html](https://lists.gnu.org/archive/html/bug-libtool/2011-09/msg00005.html)
* [https://sourceforge.net/p/libint/tickets/14/](https://sourceforge.net/p/libint/tickets/14/)
* [https://github.com/fangism/hackt/issues/21](https://github.com/fangism/hackt/issues/21)

From the other posts showing the same errors as you are observing, even though bash is installed, it is not what libtool appears to be using.  Hopefully after you read through the links it will make better sense.

Also, here is another instance of the same kind of errors due to the same underlying cause: 

- [https://lists.gnu.org/archive/html/bug-libtool/2007-12/msg00016.html](https://lists.gnu.org/archive/html/bug-libtool/2007-12/msg00016.html)

-Mark

mcsauder, 2017-12-04:
Hi @ktphys ,

Do you see the same errors if you run

`make SHELL=/bin/bash`

(Do this after a reboot if you haven't rebooted recently with everything we've attempted.)

Thanks.

-Mark

mcsauder, 2017-12-04:
Also, please copy/paste the terminal output to this post after you run

`libtool --help`

Thanks!

ktphys, 2017-12-04:
Hello Mark,

 

the error somehow changes; here’s the output:

 

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $ make SHELL=/bin/bash

make  all-recursive

make[1]: Entering directory '/usr/src/libserial-0.6.0rc2'

Making all in src

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/src'

/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c -o SerialStreamBuf.lo SerialStreamBuf.cc

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c SerialStreamBuf.cc  -fPIC -DPIC -o .libs/SerialStreamBuf.o

SerialStreamBuf.cc:1389:1: fatal error: opening dependency file .deps/SerialStreamBuf.Tpo: Permission denied

}

^

compilation terminated.

Makefile:445: recipe for target 'SerialStreamBuf.lo' failed

make[2]: *** [SerialStreamBuf.lo] Error 1

make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/src'

Makefile:453: recipe for target 'all-recursive' failed

make[1]: *** [all-recursive] Error 1

make[1]: Leaving directory '/usr/src/libserial-0.6.0rc2'

Makefile:357: recipe for target 'all' failed

make: *** [all] Error 2

 

 

Here is the output of libtool –help:

 

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $ libtool --help

Usage: libtool [OPTION]... [MODE-ARG]...

 

Provide generalized library-building support services.

 

      --config             show all configuration variables

      --debug              enable verbose shell tracing

  -n, --dry-run            display commands without modifying any files

      --features           display basic configuration information and exit

      --mode=MODE          use operation mode MODE

      --preserve-dup-deps  don't remove duplicate dependency libraries

      --quiet, --silent    don't print informational messages

      --no-quiet, --no-silent

                           print informational messages (default)

      --no-warn            don't display warning messages

      --tag=TAG            use configuration variables from tag TAG

  -v, --verbose            print more informational messages than default

      --no-verbose         don't print the extra informational messages

      --version            print version information

  -h, --help, --help-all   print short, long, or detailed help message

 

MODE must be one of the following:

 

        clean              remove files from the build directory

        compile            compile a source file into a libtool object

        execute            automatically set library path, then run a program

        finish             complete the installation of libtool libraries

        install            install libraries or executables

        link               create a library or an executable

        uninstall          remove libraries from an installed directory

 

MODE-ARGS vary depending on the MODE.  When passed as first option,

`--mode=MODE' may be abbreviated as `MODE' or a unique abbreviation of that.

Try `libtool --help --mode=MODE' for a more detailed description of MODE.

 

When reporting a bug, please describe a test case to reproduce it and

include the following information:

 

        host-triplet:   arm-unknown-linux-gnueabihf

        shell:          /bin/bash

        compiler:               gcc

        compiler flags:         -g -O2 -fstack-protector-strong -Wformat -Werror=format-security

        linker:         /usr/bin/ld (gnu? yes)

        libtool:        (GNU libtool) 2.4.2 Debian-2.4.2-1.11

        automake:

        autoconf:

 

Report bugs to <bug-libtool@gnu.org>.

GNU libtool home page: <http://www.gnu.org/software/libtool/>.

General help using GNU software: <http://www.gnu.org/gethelp/>.

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $

 

Best regards

 

Klaus

mcsauder, 2017-12-04:
Hi @ktphys ,

Thanks.  That's puzzling.  Try running it again with sudo to see if it gets farther with permissions.

It looks like your libtool shell and compiler are set correctly.  This is really a tough nut to crack! :)

I'm at a bit of a loss about what to try next with make now that we know libtool has the right shell, (other than "sudo make SHELL=/bin/bash , of course) .

Does the compile.sh script work for you?  `./compile.sh`

-Mark

ktphys, 2017-12-04:
Hi Mark,

 

with „sudo“ it’s getting much farther, but hits still on something unresolved:

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $ sudo make SHELL=/bin/bash

make  all-recursive

make[1]: Entering directory '/usr/src/libserial-0.6.0rc2'

Making all in src

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/src'

/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c -o SerialStreamBuf.lo SerialStreamBuf.cc

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c SerialStreamBuf.cc  -fPIC -DPIC -o .libs/SerialStreamBuf.o

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT SerialStreamBuf.lo -MD -MP -MF .deps/SerialStreamBuf.Tpo -c SerialStreamBuf.cc -o SerialStreamBuf.o >/dev/null 2>&1

mv -f .deps/SerialStreamBuf.Tpo .deps/SerialStreamBuf.Plo

/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialStream.lo -MD -MP -MF .deps/SerialStream.Tpo -c -o SerialStream.lo SerialStream.cc

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT SerialStream.lo -MD -MP -MF .deps/SerialStream.Tpo -c SerialStream.cc  -fPIC -DPIC -o .libs/SerialStream.o

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT SerialStream.lo -MD -MP -MF .deps/SerialStream.Tpo -c SerialStream.cc -o SerialStream.o >/dev/null 2>&1

mv -f .deps/SerialStream.Tpo .deps/SerialStream.Plo

/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT SerialPort.lo -MD -MP -MF .deps/SerialPort.Tpo -c -o SerialPort.lo SerialPort.cpp

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT SerialPort.lo -MD -MP -MF .deps/SerialPort.Tpo -c SerialPort.cpp  -fPIC -DPIC -o .libs/SerialPort.o

SerialPort.cpp: In constructor 'SerialPort::SerialPortImpl::SerialPortImpl(const string&)':

SerialPort.cpp:611:1: warning: 'SerialPort::SerialPortImpl::shadowInputBuffer' should be initialized in the member initialization list [-Weffc++]

SerialPort::SerialPortImpl::SerialPortImpl( const std::string& serialPortName ) :

^

SerialPort.cpp:611:1: warning: 'SerialPort::SerialPortImpl::queue_mutex' should be initialized in the member initialization list [-Weffc++]

SerialPort.cpp:611:1: warning: 'SerialPort::SerialPortImpl::queueDataAvailable' should be initialized in the member initialization list [-Weffc++]

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT SerialPort.lo -MD -MP -MF .deps/SerialPort.Tpo -c SerialPort.cpp -o SerialPort.o >/dev/null 2>&1

mv -f .deps/SerialPort.Tpo .deps/SerialPort.Plo

/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..    -Weffc++ -g -O2 -MT PosixSignalDispatcher.lo -MD -MP -MF .deps/PosixSignalDispatcher.Tpo -c -o PosixSignalDispatcher.lo PosixSignalDispatcher.cpp

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT PosixSignalDispatcher.lo -MD -MP -MF .deps/PosixSignalDispatcher.Tpo -c PosixSignalDispatcher.cpp  -fPIC -DPIC -o .libs/PosixSignalDispatcher.o

libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -Weffc++ -g -O2 -MT PosixSignalDispatcher.lo -MD -MP -MF .deps/PosixSignalDispatcher.Tpo -c PosixSignalDispatcher.cpp -o PosixSignalDispatcher.o >/dev/null 2>&1

mv -f .deps/PosixSignalDispatcher.Tpo .deps/PosixSignalDispatcher.Plo

/bin/bash ../libtool  --tag=CXX   --mode=link g++ -Weffc++ -g -O2   -o libserial.la -rpath /usr/local/lib SerialStreamBuf.lo SerialStream.lo SerialPort.lo PosixSignalDispatcher.lo

libtool: link: g++  -fPIC -DPIC -shared -nostdlib /usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crti.o /usr/lib/gcc/arm-linux-gnueabihf/4.9/crtbeginS.o  .libs/SerialStreamBuf.o .libs/SerialStream.o .libs/SerialPort.o .libs/PosixSignalDispatcher.o   -L/usr/lib/gcc/arm-linux-gnueabihf/4.9 -L/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf -L/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../.. -L/lib/arm-linux-gnueabihf -L/usr/lib/arm-linux-gnueabihf -lstdc++ -lm -lc -lgcc_s /usr/lib/gcc/arm-linux-gnueabihf/4.9/crtendS.o /usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crtn.o  -O2   -Wl,-soname -Wl,libserial.so.0 -o .libs/libserial.so.0.0.0

libtool: link: (cd ".libs" && rm -f "libserial.so.0" && ln -s "libserial.so.0.0.0" "libserial.so.0")

libtool: link: (cd ".libs" && rm -f "libserial.so" && ln -s "libserial.so.0.0.0" "libserial.so")

libtool: link: ar cru .libs/libserial.a  SerialStreamBuf.o SerialStream.o SerialPort.o PosixSignalDispatcher.o

[... 66 lines cut ...]
make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/sip'

Makefile:453: recipe for target 'all-recursive' failed

make[1]: *** [all-recursive] Error 1

make[1]: Leaving directory '/usr/src/libserial-0.6.0rc2'

Makefile:357: recipe for target 'all' failed

make: *** [all] Error 2

pi@raspberryKT:/usr

 

I find no script “./compile.sh”.

 

Best regards

 

Klaus

mcsauder, 2017-12-04:
Hi @ktphys ,

That's great!  It was able to compile everything but the python code successfully.  This means that libtool was choosing the wrong shell, although I have absolutely no idea why the permissions error showed up.

The reason it is failing in the sip directory is likely because you need to install python-sip

`sudo apt install python-sip-dev`

mcsauder, 2017-12-04:
Hi again @ktphys ,

The file compile.sh is in your top level libserial directory, it is a bash script that simply contains instructions you can run yourself from the terminal:

`mkdir -p build`
`cd build`
`cmake ..`
`cmake -G Sublime\ Text\ 2\ -\ Unix\ Makefiles .. `
`cd ..`
`make -j3 -C build`

You can run the script by `./compile.sh`

ktphys, 2017-12-04:
Hi,

 

Now make was running:

 

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $ sudo make SHELL=/bin/bash

make  all-recursive

make[1]: Entering directory '/usr/src/libserial-0.6.0rc2'

Making all in src

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/src'

make[2]: Nothing to be done for 'all'.

make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/src'

Making all in doc

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/doc'

Making all in UML

make[3]: Entering directory '/usr/src/libserial-0.6.0rc2/doc/UML'

make[3]: Nothing to be done for 'all'.

make[3]: Leaving directory '/usr/src/libserial-0.6.0rc2/doc/UML'

make[3]: Entering directory '/usr/src/libserial-0.6.0rc2/doc'

make[3]: Nothing to be done for 'all-am'.

make[3]: Leaving directory '/usr/src/libserial-0.6.0rc2/doc'

make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/doc'

Making all in examples

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/examples'

make[2]: Nothing to be done for 'all'.

make[2]: Leaving directory '/usr/src/libserial-0.6.0rc2/examples'

Making all in sip

make[2]: Entering directory '/usr/src/libserial-0.6.0rc2/sip'

python ./configure.py

2017-12-04 21:23:21,664 INFO Build file is libserial.sbf

2017-12-04 21:23:21,664 INFO Reading sip configuration.

2017-12-04 21:23:21,665 INFO Running sip command: '/usr/bin/sip -c . -e -b libserial.sbf libserial.sip'

sip: Deprecation warning: libserial.sip:1: %Module version number should be specified using the 'version' argument

make -f Makefile.sip

make[3]: Entering directory '/usr/src/libserial-0.6.0rc2/sip'

g++ -c -g -O2 -fstack-protector-strong -Wformat -Werror=format-security  -D_FORTIFY_SOURCE=2 -fPIC -Wall -W -D_REENTRANT -DNDEBUG -I. -I../src -I/usr/include/python2.7 -o siplibserialcmodule.o siplibserialcmodule.cpp

[... 188 lines cut ...]
config.h      depcomp        libserial.pc.in    Makefile.dist

config.h.in   doc            libserial.spec     Makefile.in

config.log    doxygen.conf   libserial.spec.in  missing

 

 

So, after all, what is the resume? I’m a bit confused ….

 

Which steps have to be followed If somebody needs to install everything from scratch?

Is it possible to amend the auto-generated “Makefile” appropriately?

 

Thank you very much for your constant support !!!!

 

Klaus

mcsauder, 2017-12-04:
Hi @ktphys ,

[Here](https://github.com/crayzeewulf/libserial/blob/master/compile.sh) is a link to the CMake compile script file.

I would appear that you have everything installed properly at this point and should be able to use LibSerial.  If you were only interested in the current release, there is a release package that can be installed using apt:

`sudo apt install libserial-dev`

which will give you the same functionality as the [master](https://github.com/crayzeewulf/libserial) branch of the git repo minus some documentation updates.  We are working toward a new release candidate from the [c++14](https://github.com/crayzeewulf/libserial/tree/c++14) branch that eventually will be ableto be installed from apt as well.

Your case was pretty tricky, and although I have updated the documentation to include the additional packages your installation required, I'm not sure how to enforce this through the makefiles.  Would you be willing to try inserting the line 

`SHELL=/bin/bash`

into first the configure.ac file, and then the Makefile.am at the top level to see if it allows you success?  If you can test a way to get this into the make files that functions we would be happy to include that in the repo.

Thanks for your work on this, and thanks for reporting the issue.

Let me know once you would like to close the issue!

-Mark

ktphys, 2017-12-05:
Hello Mark,

 

I did not find the configure.ac file on top level (and also not below): 

 

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $ ls

ABOUT-NLS     config.log     doxygen.conf     libserial.spec.in  missing

aclocal.m4    config.status  doxygen.conf.in  libtool            NEWS

AUTHORS       config.sub     examples         ltmain.sh          README

ChangeLog     configure      INSTALL          m4                 sip

compile.sh    configure.in   install-sh       Makefile           src

config.guess  COPYING        libserial.pc     Makefile.am        stamp-h1

config.h      depcomp        libserial.pc.in  Makefile.dist

config.h.in   doc            libserial.spec   Makefile.in

 

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $

 

I altered the Makefile.am, and tried to re-run “configure”:

 

sudo ./configure

sudo: ./configure: command not found               --->????????

 

Rather strange! “./configure”  is existing!

pi@raspberryKT:/usr/src/libserial-0.6.0rc2 $ ls -al configure

-rw-r--r-- 1 pi pi 556498 Jun  5  2014 configure

 

For some reason I have to run it this way:

 

sudo sh ./configure     --- not clear to me why!

 

Now “sudo make” works correctly!

(also “sudo make install” and “sudo make clean”)

 

Best regards

Klaus

mcsauder, 2017-12-05:
Hi @ktphys ,

[Here](https://github.com/crayzeewulf/libserial/blob/master/configure.ac) is the file in the repo.  I do not know why you are missing some of the files in the repo.  It seems like you might be best served by checking out a fresh clone of the repository.

Please open a terminal, navigate to your libserial directory, enter the list command, `ls`, and copy/paste the terminal output back here.  Once you have inspected the list of available files, compare it to the list of files on the [repo](https://github.com/crayzeewulf/libserial).  If you are missing any of the files at all, (even one), you need to fix your repo.  If you are not missing any of the files, please locate configure.ac and compile.sh and work on the previous instructions we discussed.

Thanks!

mcsauder, 2017-12-05:
Aha!  I see why you don't have some of the files, you didn't clone the repo, you downloaded the release candidate package.

Please clone the repo and work with the latest code.

-Mark

ktphys, 2017-12-05:
Hello Mark,

 

I downloaded a few days ago „libserial-0.6.0rc2.tar.gz”.

Now I copied the actual repo and will try that one tomorrow.

 

Best regards

Klaus

mcsauder, 2017-12-19:
Hi @ktphys ,

Any news?  How are things going?

mcsauder, 2018-01-15:
Hi @ktphys ,

I hope things are working for you.  I am going to close this issue under the assumption that you are able to build with the current code.  If this is not the case please feel free to re-open the issue.

Thanks again for reporting this issue!

-Mark

closed 2018-01-15 by mcsauder