crayzeewulf/libserial #99: SerialPort and std::this_thread::sleep_for(...)

josechenf, 2017-12-20:
Hello!

I am using C++11 and the SerialPort class.

I noticed that my program wasn't sleeping when I used std::this_thread::sleep_for(...) after calling serial_port.Write(data_buffer). 
It is somewhat strange, as if I do two consecutive std::this_thread::sleep_for(...)'s that are long enough (e.g. 5 seconds), the first one doesn't sleep but the second one does. 

If it helps at all, I am making an std::shared_ptr of a SerialPort object, and then calling a few consecutive serial_port->Write(data_buffer) from within different class objects. If I comment the lines with serial_port->Write(data_buffer) it sleeps as expected.

Is there something I am doing wrong? Should I be emptying the buffer between calls or waiting in between writes? (Is there a function to wait until writing is done?)

Thank you! (Great library btw)

mcsauder, 2017-12-20:
Hi @josechenf ,

Thanks for posting this issue!  I have been able to reproduce what you are seeing, which is great, except that I have not the first clue about why it is happening.  I also have a suspicion that the issue lies deeper than LibSerial.

However, your question about a function for waiting until writing is done has created what appears to be a successful workaround.  I've added a "DrainWriteBuffer()" method to wrap the tcdrain() system method which will accomplish what you've requested and by utilizing that method after writing to the serial port it appears to avoid encountering your issue.

I reproduced your issue in the multi-thread SerialStream and SerialPort tests in UnitTests.cpp.  By commenting out lines 1784, 1828, 1874, and 1923 in UnitTests.cpp in PR #97 you can see the issue you've reported actually occurring.  In the testing I've done I haven't encountered the issue as long as DrainWriteBuffer() is called, although I've noticed that the issue doesn't happen with certainty when it does occur, so I'm not sure if I've simply reduced the probability of encountering it or if it is actually being avoided.

I must admit freely that I really have no clue what is causing the issue, but I cannot find anything in LibSerial that should be impacting the behavior of usleep().

Please let us know if you have additional insights or suggestions as this feels like a problem with lower level system calls.

Thanks again for reporting the issue!

-Mark

josechenf, 2017-12-21:
@mcsauder 
Thanks, that seems to have fixed it. I haven't encountered the issue again!

closed 2017-12-21 by josechenf