crayzeewulf/libserial #107: License issue: COPYING.LESSER contains text of LGPL-3.0 instead of LGPL-2.0

ffontaine, 2018-06-04:
COPYING.LESSER contains the license of LGPL-3.0 but source files contain license of LGPL-2.0. Could you confirm that libserial is licensed under LGPL-2.0+? In this case, could you replace COPYING.LESSER with the text of LGPL-2.0?

ffontaine, 2018-06-05:
MIT is the most popular license on github.

MIT and BSD-2-Clause permits use, redistribution, redistribution with modification and contains provisions to retain the copyright notice and warranty disclaimer.
In addition, MIT explicitly allows sub-licensing, selling, publishing and merging.

On his side, BSD-3 Clause adds a non-endorsment clause.

It should be noted that these three licenses don't have any patent license grant clause. If you want that, Apache-2 is probably the way-to-go.

It should be noted that for a license change, you'll have to get the agreements of **all** contributors.

crayzeewulf, 2018-06-05:
@mcsauder, I am completely in favor of any license that allows more usage without restrictions (including commercial use without any restrictions). So either BSD or MIT would be okay with me.

mcsauder, 2018-06-07:
@wedesoft and @tpetazzoni , 

Given the discussion above, what are your thoughts?  Thanks!

-Mark

tpetazzoni, 2018-06-10:
@mcsauder I'm not sure why my opinion matters. However, as @ffontaine explained, doing a license change requires the agreement of all contributors who have ever contributed to the project. Perhaps this is an effort that should be done separately from fixing the current existing license file ?

mcsauder, 2018-06-12:
Thanks @tpetazzoni , I appreciate the feedback. To your point, I've created PR #108 to deal directly with this issue in isolation and will create a separate issue and pull request to address migrating the license.

@ffontaine, let me know if PR #108 addresses your concern.

Thanks!

-Mark

ffontaine, 2018-06-12:
PR #108 seems good to me actually the issue was first spotted by @tpetazzoni when I bumped the version of libserial in buildroot.
Could you merge this PR and release a new version of libserial?

Thanks,

Fabrice