crayzeewulf/libserial #109: [Feature] Migrate LibSerial License from LGPL to more permissive license.

mcsauder, 2018-06-12:
Hi CrayzeeWulf and libserial contributors, (@wedesoft, @tpetazzoni, @ffontaine),

I would like to propose migrating LibSerial to a more permissive license.  The rationale behind this proposal is to promote more widespread adaption at a commercial level to also facilitate more widespread contribution back into this library.

It is my opinion that the library may achieve a broader user base if the license is migrated to a commercially permissive license, and I would propose focusing primarily on one of the following license options: 1) the BSD 2-Clause license, 2) the BSD 3-Clause, 3) the MIT license, or 4) Apache-2.

While the LGPL is not unfriendly to commercial usage, there remains much concern in the business realm to allow the usage of any code that is [GPL](https://en.wikipedia.org/wiki/GNU_General_Public_License)-related, which unfortunately also includes the [LGPL](https://www.gnu.org/licenses/old-licenses/lgpl-2.0.en.html).  This is due to the (intended) [copy-left](https://www.gnu.org/licenses/copyleft.en.html) nature of the GPL, and the related nature of the LGPL.

While I do understand the intentions behind the GPL and remain supportive of that initiative, commercial users are often important contributors to many open source projects under more permissive licenses.

I am creating this issue to facilitate and archive discussion regarding the effects and opinions regarding a license change of this library.

To anyone watching LibSerial with an opinion to share, please share your thoughts!

Thanks!

-Mark

crayzeewulf, 2018-06-13:
Mark:

Thanks for initiating this conversation. As I commented in #107, it will be good to move LibSerial to a more permissive license. After looking through the various licenses, I am personally leaning towards BSD 3-Clause mainly due to the non-endorsement clause included with it. The four licenses in order of my preference (high to low):

1. BSD 3-Clause
1. BSD 2-Clause
1. MIT
1. Apache-2

Of course, additional discussion with the rest of the developers might change my mind but I wanted to kick things off with my list.

mcsauder, 2018-10-12:
Closing with PR #115. Thanks Crayzee Wulf!

closed 2018-10-12 by mcsauder