crayzeewulf/libserial #110: [FEATURE] LibSerial base class

mcsauder, 2018-06-13:
Hi,

Prior to a v1.0.0 release candidate, it would be nice to create a parent/base class for SerialPort, SerialStreamBuf, and SerialStream to inherit common methods from.

I would like to make another attempt to resolve this inheritance structure prior to the release of v1.0.0 and appreciate any suggestions on naming or convention toward this effort!

Thanks!

-Mark

crayzeewulf, 2018-09-14:
Mark:

Just to give some context, my intention was to have `SerialStreamBuf` and `SerialStream` work the same way as [std::streambuf](https://en.cppreference.com/w/cpp/io/basic_streambuf) and [std::iostream](https://en.cppreference.com/w/cpp/io/basic_iostream) work. So, they have the same relationship as the corresponding classes in the standard library. It allows users to use all standard library functions that work on [std::iostream](https://en.cppreference.com/w/cpp/io/basic_iostream) with `SerialStream`.

`SerialPort`, on the other hand, was intended to be a standalone class that could be used by itself on some embedded platforms that did not support the C++ standard library back when `LibSerial` started. There were several C++ compilers for single-board computers in those days that did not include the C++ standard library and/or did not support all C++ features. This is not a concern any more and compilers have come a long way. So, your suggestion to reduce code repetition between these classes makes sense. It will simplify the overall code. I do not have any suggestions for the hierarchy or naming at this moment but will chime in with more shortly.

mcsauder, 2018-10-11:
Thanks for doing this work @crayzeewulf !  Closing with your commit https://github.com/crayzeewulf/libserial/commit/68d0b3701db0f65c5dc6e247c89ccb49da2ad38a

closed 2018-10-11 by mcsauder