crayzeewulf/libserial #111: Please help me regarding libserial library

saipothanjanjanam, 2018-07-06:
Dear crayzeewulf,

[image]
[image]

Please I am facing this problem when executing the program.
Program is compiling good but it is not executing.
Please help me with this issue.

Thank you,
With my best regards,
Sai pothan

mcsauder, 2018-07-06:
Hello @saipothanjanjanam ,

If you have cloned the current master branch of the repo and are building it, your issue could be occurring because the debian package of LibSerial is out of step with the Girhub repo.  

First, could you use apt to remove LibSerial,

`sudo apt remove libserial-dev`

Next, use the make instructions to install the library built from your local repo clone, (the make instructions in the README, rather than the compile script which uses CMake):

```
make -f Makefile.dist
./configure --prefix=/usr/include/
make
sudo make install
```

Last, recompile your program and let us know if the issue persists.  If the issue does remain, let us know and I'll dig into it further with you!

-Mark

saipothanjanjanam, 2018-07-06:
Dear Mark,

Thank you very much. I have used locate and forcedly delete all the
libserial files. Then,  installed like sudo apt-get install libserial-dev.
It is working now.

Best Regards,
Sai pothan

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

mcsauder, 2018-07-07:
Glad to hear things are fixed up!

closed 2018-07-07 by mcsauder