crayzeewulf/libserial #113: No documentation on how to link against the library

CharlesB2, 2018-09-17:
It would be very useful to add this to the doc or tutorial, there's a lot on how to build the library itself, but when it comes to compile a program I couldn't find how.

CharlesB2, 2018-09-20:
I think you can directly tell in the docs or README that you can either link against libserial_static, or call use pkg_config, since you have enabled the use of it (and it's very convenient and portable)

FisherTiger95, 2018-09-21:
Hi, I did not understand how to link the library to a program, I added the libserial_static to target_link_libraries, but i get the error:
`/usr/bin/ld: cannot find -llibserial_static`
Even if I followed all the instructions to install the library, did I miss something?

mcsauder, 2018-10-16:
Hi @CharlesB2 ,

We have a standalone project in the dev branch using CMake [here](https://github.com/crayzeewulf/libserial/tree/dev/examples/example_project), and references in the README.md that give a copy/paste project template.

Let us know if you like the outcome and if we can close out this issue!

-Mark

CharlesB2, 2018-10-17:
Great! If you can raise CMake minimum version to 3.8, we can greatly simplify the CMakeLists file since it has support for threads and c++11 requirements. If you agree I'll set up a PR for this.

CharlesB2, 2018-10-19:
Started something, but I hit an issue: users must `#include <libserial/SerialPort.h>` but these `#include "SerialConstants.h"`, which is not found because it doesn't prefix with `libserial/`?

crayzeewulf, 2018-10-19:
@CharlesB2 Thanks for reporting this. This is likely a bug that needs to be fixed. Recently, we started transitioning the headers so that they are installed under "libserial" folder to avoid conflicts with other packages. However, it appears like not all have been transitioned correctly. I do not think I communicated this very well to Mark either. We will get this fixed shortly. I am not going to be able to work on this over the weekend but will update with a fix on Monday.

crayzeewulf, 2018-10-23:
@CharlesB2 I've just pushed a commit (62b46da) that should fix the header issue. All LibSerial headers ought to be included as `#include <libserial/xyz.h>` (or `#include "libserial/xyz.h"` in internal `.cpp` files) now. 

One thing still missing the example CMake project is the ability to find LibSerial using [find_package()](https://cmake.org/cmake/help/latest/command/find_package.html). I will work on implementing this over the next few days.

crayzeewulf, 2018-10-23:
Perfect! Thanks! I will test it out in my setup and update the example. However, if you like, please feel free to submit a PR as Mark had suggested.

abrock, 2019-05-06:
> ```
> find_package(PkgConfig)
> pkg_check_modules(SERIAL libserial)
> 
> add_executable(test main.cpp)
> target_link_libraries(test PRIVATE ${SERIAL_LIBRARIES} ${CMAKE_THREAD_LIBS_INIT})
> ```

I tried that on Debian testing with libserial installed via apt ("libserial | 0.6.0~rc2+svn122-4") and linking fails:

```error: /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/libserial.so: undefined reference to `pthread_mutex_trylock'```

Everything works when I add "pthread" to the list of libraries to link against, therefore I think the docs should mention that.

crayzeewulf, 2020-08-09:
This has been resolved through the example project and pkg-config support in libserial. Closing but if there is still an outstanding issue please let us know.

closed 2020-08-09 by crayzeewulf