crayzeewulf/libserial #121: Missing stdexcept?

cjvaughter, 2018-10-30:
I can't build without adding `#include <stdexcept>` to SerialPortConstants.h
Am I missing something?

crayzeewulf, 2018-10-31:
@cjvaughter Could you please post the Linux distribution and compiler version you are using?

cjvaughter, 2018-10-31:
Distro: pipaOS (Raspbian-based)
Compiler: g++ 4.9.2
Branch: master

Everything works fine after adding the header. To be honest, I'm not really sure how it could compile without it.

mcsauder, 2018-10-31:
Hi @cjvaughter , the dev branch contains that include, so this issue will be resolved as soon as the next dev->master PR is created/accepted.

Would you be willing to checkout the dev branch and verify that it corrects the issue you have identified?

Thanks!