crayzeewulf/libserial #124: serial.h THROW macro conflicts with Windows ATL THROW macro

remrol, 2018-11-29:
Under Windows, if I include serial.h in my cpp file which includes ATL headers then I get following warning:
warning C4005: 'THROW': macro redefinition
1>c:\program files (x86)\microsoft visual studio\2017\professional\vc\tools\msvc\14.16.27023\atlmfc\include\afx.h(858): note: see previous definition of 'THROW'
suggestions:
1 Rename THROW macro to e.g. SERIAL_THROW or similar.
2 Move THROW macro from serial.h to one of the impl files. In fact there's no THROW in serial.h, only in impl files.

crayzeewulf, 2018-11-29:
@remrol I am confused by your question. Which `serial.h` file are you referring to? There is no `serial.h` file in this project and no `THROW` macro. Furthermore, `libserial` does not work on Windows; it is mainly for Linux operating system. Maybe you meant to add this issue to some other project?

crayzeewulf, 2018-11-29:
Perhaps this issue is for [wjwwood/serial](https://github.com/wjwwood/serial) project?

remrol, 2018-11-29:
Yes I am really sorry I screwed projects, sorry once again !!!