crayzeewulf/libserial #129: serial_port_read_write builds but program aborts( with core dump)

gazzatav, 2019-02-19:
I'm not an expert in serial comms but recently discovered this library and it seems just what I need for some work involving modbus.  First, thanks for your work on this crayzeewulf.  Wanting to try something out straight away I built the example serial_port_read_write.cpp (from examples) with:
 `g++ serial_port_read_write.cpp ../src/SerialPort.cpp -I../src` on the command lline.  I changed /dev/ttyUSB0 and /dev/ttyUSB1 to /dev/ttyS0 and /dev/ttyS1 as these were present in /dev and the others were not.  When I ran the program it aborted with the message:
terminate called after throwing an instance of 'LibSerial::OpenFailed'
  what():  Bad file descriptor
Aborted (core dumped)

I'm a bit puzzled about how these ports could even talk to each other with out something else going on in the background to 'tie' them.  I used socat to create byte streams between the ports but when I tried again the same error occurred.  What am I missing?

mcsauder, 2019-02-19:
Hi @gazzatav , I am now noticing that this isn't in the README and will update it if this fixes your issue... Assuming you are running on a linux platform, have you given yourself permissions to run those ports and (you will need to reboot your computer after the permissions are applied):
```
sudo usermod -a -G dialout $USER
sudo usermod -a -G plugdev $USER
```

Try that, and let us know if it fixes anything for you.

mcsauder, 2019-02-19:
@gazzatav , I've opened PR#130, if you spot any other places we could help this example code out please feel free to comment in that PR!

gazzatav, 2019-02-20:
Thanks for the replies.  I have the program running now (on Ubuntu 18.04).  I was running the example and socat as sudo so there should be no problem with groups but that is useful information to have in the README.  I believe the invocation of socat which I used originally was the problem and it may be good to have an example of one that works in the README.  The one below worked for me.

`socat -d -d pty,raw,echo=0,link=/dev/ttyS0 pty,raw,echo=0,link=/dev/ttyS1`

My original socat invocation just gave it the filenames as addresses, I thought that would create streams between the ports, perhaps I did not give sufficient options - I don't know enough about socat.

mcsauder, 2019-02-20:
@gazzatav , I've submitted PR #132 with the socat instructions you gave us.  Once that PR is accepted, if it is enough to close this issue please feel free to do so.  Otherwise just let us know what else you have in mind!  Thanks for submitting this issue!

-Mark

closed 2019-02-21 by gazzatav