crayzeewulf/libserial #134: Fedora 29: can not find PYTHON_LIBRARIES when trying to compile

onkami, 2019-03-09:
Tried latest releases and development, and received the following error upon running ./compile.sh:

CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindPythonLibs.cmake:265 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:16 (FIND_PACKAGE)

How I could get it fixed or worked around? What libraries it might want? Many python libraries are already installed in the system.

crayzeewulf, 2019-03-09:
@onkami, could you please install `python3-devel` package and try again? You can use the following command to install it:

```sh
sudo dnf install python3-devel
```

onkami, 2019-03-09:
Thank you it helped !

closed 2019-03-09 by onkami