crayzeewulf/libserial #135: Remote device resets, leaving Libserial instance unusable

tykbry, 2019-03-25:
I have a serial port device over a usb, on a rasberry linux environment. 

If no data have been transmitted in a given time frame, the device resets (effectively shutting down and starting again). This has the consequence that the underlying file descriptor becomes invalid.

Now I can detect the resulting EIO (Input/output error exeption) from attempts to read or write, but the problem is I have no way of resetting the Libserial serial port. An attempt to close the port will result in the same error, because it will try to set the termios to default. In addition this prevents the object getting deleted (it will try to close and fail -> terminate).
And I can't open it again either because the file descriptor is still it's old value and it thinks the connection is open.

TL:DR 
Device resets, leaving the object in an unusable state, that can't used, reset, deleted or really anything

crayzeewulf, 2019-03-25:
@tykbry Is LibSerial throwing an exception when you encounter this issue?

I've had to work with similar devices in the past and I ended up using a separate thread to periodically "ping" the device to prevent it from disconnecting or resetting. Is this an option in your application?

crayzeewulf, 2019-03-25:
@tykbry Thanks for bring this to our attention. It looks like by calling `SerialPort::Close()` the destructor of `SerialPort` class is violating the rule that [destructors should not throw any exceptions](https://isocpp.org/wiki/faq/exceptions#dtors-shouldnt-throw) (or more precisely, no exceptions should escape destructors). This needs to be fixed in LibSerial. I will work on updating LibSerial to fix this issue and update the code. Hopefully, that will help find a solution to your issue. I hope to have this fixed in the evening today.

tykbry, 2019-03-25:
@crayzeewulf Thanks for being so fast.
 It is my plan to have a thread pinging it periodically. But if for some reason it fails for the given time period, I have no way to gracefully recover. I could make a new SerialPort and not delete the old one (memory leak) or make it run in its own process and restart that.

tykbry, 2019-03-25:
And any error would be bad, as it need to run headless. So no to little administration.

crayzeewulf, 2019-03-25:
@tykbry You are correct. This needs to be fixed in LibSerial. You ought to be able to delete/destroy an instance of `SerialPort` no matter what without it throwing an exception or terminating the application.

crayzeewulf, 2019-03-26:
@tykbry I've submitted a potential fix for this issue in the `fix_issue_135` branch. I tested it by running the following sample code that continuously sends data to a serial port and recreates the `SerialPort` object in case of an error. I physically unplugged and re-plugged a USB-Serial converter while this example was running and it continued to run without terminating. Will this approach work for your project? Please give this branch a try and let us know what you observe.

```c++
#include <libserial/SerialPort.h>
#include <libserial/SerialStream.h>

#include <iostream>
#include <chrono>
#include <thread>

int main()
{
    using LibSerial::SerialPort ;
    using namespace std::literals ;

    while(true)
    {
        try {
            SerialPort serial_port ;
            // Open hardware serial ports using the Open() method.
            std::cout << "Opening serial port." << std::endl ;
            serial_port.Open("/dev/ttyUSB0") ;

            // Set the baud rates.
            using LibSerial::BaudRate ;
            serial_port.SetBaudRate(BaudRate::BAUD_115200) ;

            // Remove the serial port hardware while this loop is running
            // to see if we can recover.
            for(int i=0; i<100; ++i)
            {
                std::cout << "." << std::flush ;
                serial_port.WriteByte('a') ;
                std::this_thread::sleep_for(500ms) ;
            }

            std::cout << "Closing serial port." << std::endl ;
            serial_port.Close() ;
        } catch(const std::exception& err) {
            std::cerr << "Communication failed. Retrying in 1s." << std::endl ;
            std::this_thread::sleep_for(1s) ;
        }
    }
    return EXIT_SUCCESS ;
}
```

tykbry, 2019-03-26:
@crayzeewulf Works for me. I can now catch the error, and restart (close and reopen) the connection.

crayzeewulf, 2019-03-26:
@tykbry Thanks for checking and reporting back. I am glad to hear that it worked for you. I will merge the changes in the `master` branch in the evening today.

closed 2019-04-24 by mcsauder