crayzeewulf/libserial #137: Build error for C++11 (default in CMakeLists.txt)

kebrahimpour, 2019-04-24:
build fails when  I run compile.sh
**Error message:**
libserial/test/SerialPortUnitTests.cpp:855:26: error: ‘std::literals’ has not been declared
     using namespace std::literals::chrono_literals ;
                        ^
compilation terminated due to -Wfatal-errors.
**reason:** Compiler is set to C++11 in CMakeLists.txt 

we **need C++14** for std::literals

mcsauder, 2019-04-24:
Hi @kebrahimpour , thanks for reporting this issue.  Which OS distribution are you running?  Does the addition of `using namespace std::literals;` at line ~44 of `SerialPortUnitTests.cpp` resolve the issue for you by chance?

kebrahimpour, 2019-04-25:
> Hi @kebrahimpour , thanks for reporting this issue. Which OS distribution are you running? Does the addition of `using namespace std::literals;` at line ~44 of `SerialPortUnitTests.cpp` resolve the issue for you by chance?
hi
1. I am working on a ubuntu 16.04 virtual machine

2. using std::literals didn't help. I get same error message

> /home/md/CLionProjects/Sandbox/libserial/test/SerialPortUnitTests.cpp:44:22: error: ‘literals’ is not a namespace-name
>  using namespace std::literals;
> 
3.   if I set  **CMAKE_CXX_STANDARD to 14** in cmake it works
this was the reason why I wrote that we need c++14  (to build tests).
AND accordig to the documentation Libserial needs C++14 to run/build
We have this issue only in SerialPortUnitTests.cpp

Is there any chance to make all build and run using C++11 (including tests)?

thanks

crayzeewulf, 2019-04-26:
@kebrahimpour You are absolutely right, we do need to set `CMAKE_CXX_STANDARD` to 14. It used to be there but got accidentally removed in commit aaf2e34617394fe7cdd31080171b3ebfd6bf80fc. I will add it back shortly to fix this issue.

crayzeewulf, 2019-04-26:
@kebrahimpour For some reason, one of your comments is not showing up here on Github (at least I cannot see it but I have it in my e-mail). You asked: "Is there any chance to make all build and run using C++11 (including tests)?" Unfortunately, the answer is no because the code uses several C++14 features that are not available in C++11.

mcsauder, 2019-04-27:
@crayzeewulf and @kebrahimpour , I just did a fresh install of Ubuntu 19.04, and with no new packages installed I was able to compile and run everything... gcc version is 8.3.0, but no special instructions are required. From my vantage point C++14 isn't a special requirement, however, I know that from the documentation the usage should require C++14  for `ms` suffixes...  What am I missing?

kebrahimpour, 2019-04-27:
thanks @crayzeewulf.
exactly. I am able to build the essential part  in  C++11. As I see C++14 is needed for tests too.  But when you are finished with tests and generated docs then you can rebuild the main part in C++11

thanks again

crayzeewulf, 2019-04-28:
> @crayzeewulf and @kebrahimpour , I just did a fresh install of Ubuntu 19.04, and with no new packages installed I was able to compile and run everything... gcc version is 8.3.0, but no special [...]

@mcsauder I believe that the answer is that different versions of g++ use different C++ standards as the **default**. Based on [this](https://stackoverflow.com/a/44735016/1062948) answer on SO, for example, one can determine the default version of C++ used by g++ by running the following command:

```
g++ -dM -E -x c++  /dev/null | grep -F __cplusplus
```

The results for different Ubuntu releases (and corresponding versions of g++) are listed below:

* Ubuntu 16.04 (g++ 5.4.0): `#define __cplusplus 199711L`
* Ubuntu 18.04 (g++ 7.3.0): `#define __cplusplus 201402L`
* Ubuntu 19.04 (g++ 8.3.0): `#define __cplusplus 201402L`

As seen here, Ubuntu 18.04 and 19.04 use g++ versions that use C++14 as the default standard. Whereas Ubuntu 16.04 uses a much older version of g++ for which the default C++ standard is well before C++14 (one has to explicitly select a newer C++ standard using `-std` command line option in this case). Hope that makes sense.

mcsauder, 2019-04-29:
That makes perfect sense.  Thanks for your explanation @crayzeewulf . I've been on 18.04, 18.10, and 19.04 since the PR you referenced was accepted, which is how it slipped by me at the time as well.

crayzeewulf, 2019-04-30:
@mcsauder same here; I've been using 18.04 and higher for all tests since that PR. 

I will have some time this week to fix the CMake configuration and test it out on 16.04. Hopefully, I won't let it slip by again.

crayzeewulf, 2019-05-27:
Fixed via pull request #141. Merged.

closed 2019-05-27 by crayzeewulf