crayzeewulf/libserial #139: Manual installation problem

Metalzero2, 2019-05-23:
I am writing down a few problems I had while trying to install the library using CMake

**Problem 1**
CMake was not able to download googletest 1.8.1 for some reason (even though the link is working). So I manually needed to download the file and changed the URL in the CMake file to where I locally had it.

**Problem 2**
It was not able to install "sphinx_rtd_theme", so I had to manually install that as well, by running the command $ pip install sphinx_rtd_theme.

Everything else went fine. Maybe this will help somebody else that came across the same issues.

crayzeewulf, 2019-05-23:
@Metalzero2 Thanks for reporting this issue. Could you please let us know which Linux distribution and version you are using? Maybe that will help us resolve this issue.

crayzeewulf, 2019-05-23:
Additionally, you ought to be able to do a minimal build without tests, documentation, or examples using the following commands:

```
cmake -H. -BRelease \
    -DCMAKE_BUILD_TYPE=Release \
    -DLIBSERIAL_ENABLE_TESTING:BOOL=OFF \
    -DLIBSERIAL_BUILD_EXAMPLES:BOOL=OFF \
    -DLIBSERIAL_PYTHON_ENABLE:BOOL=OFF \
    -DLIBSERIAL_BUILD_DOCS:BOOL=OFF
cmake --build Release
```

Maybe we will add these options to the top-level README file in the future.

Metalzero2, 2019-05-28:
@crayzeewulf 
My system is Ubuntu 04.18 LTS (Kubuntu flavor). Additionally, my CMake version is 3.14.1, if it helps. If you need any other information, let me know.

crayzeewulf, 2019-05-30:
Thanks for the information, @Metalzero2. I use Ubuntu 18.04 regularly for building LibSerial and have not encountered an issue with googletest so far. The commands that I typically use for building on Ubuntu 18.04 are listed in this [Dockerfile](https://github.com/crayzeewulf/libserial/blob/master/dockerfiles/ubuntu/18.04/Dockerfile). Maybe looking through that file might give you some ideas.

Regarding `sphinx_rtd_theme`, one might have to install `python-sphinx-rtd-theme` or `python3-sphinx-rtd-theme` package on Ubuntu using one of the following commands:

```sh
sudo apt install python3-sphinx-rtd-theme
sudo apt install python-sphinx-rtd-theme
```
I will document this in the top-level README file shortly.

mcsauder, 2020-01-21:
Hi @Metalzero2 , is this issue fixed for you?  Thanks for your feedback!

Metalzero2, 2020-01-22:
Hi @mcsauder . For the time being, I have already kept the installation I did. I prefer not to uninstall it now that I have it working. When I get the chance to install it again, I will let you know if there is an issue. Unstill then, fell free to close this issue :)

crayzeewulf, 2020-03-14:
Closing issue as suggested by the original author.

closed 2020-03-14 by crayzeewulf