crayzeewulf/libserial #142: Cmake integration

giellamoswhard, 2019-06-06:
The generation of a libserialConfig.cmake would ease the integration of the library.

I made a small try.
```
(6353) $ git diff HEAD~1..HEAD 
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 2633583..c212251 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -10,7 +10,7 @@ add_library(libserial_static STATIC ${LIBSERIAL_SOURCES})
# another "lib" prefix.
#
 set_target_properties(libserial_static PROPERTIES PREFIX "")
-target_include_directories(libserial_static PUBLIC ${CMAKE_CURRENT_SOURCE_DIR})
+target_include_directories(libserial_static PUBLIC $<INSTALL_INTERFACE:${CMAKE_INSTALL_INCLUDEDIR}> $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}>)
 target_link_libraries(libserial_static Threads::Threads)

 #
@@ -30,6 +30,7 @@ install(DIRECTORY libserial
 #
 if (INSTALL_STATIC)
     install(TARGETS libserial_static
+            EXPORT libserialConfig
             ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR}
             LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR}
             RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})
@@ -60,7 +61,10 @@ if (INSTALL_SHARED)
         SOVERSION ${PROJECT_VERSION_MAJOR}
     )
     install(TARGETS libserial_shared
+            EXPORT libserialConfig
             ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR}
             LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR}
             RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})
 endif()
+
+install(EXPORT libserialConfig DESTINATION ${CMAKE_INSTALL_LIBDIR}/cmake)
```

crayzeewulf, 2019-06-06:
@giellamoswhard You are correct and I've been wishing to do this for a while. You might have motivated me to finish this task. Stay tuned.

crayzeewulf, 2020-04-04:
Support for pkg-config while building using CMake has been implemented in [this](https://github.com/crayzeewulf/libserial/tree/packaging_using_cpack) branch. It will be merged into master after testing.

GravisZro, 2023-10-08:
This issue can be can be closed as it was resolved long ago.

closed 2023-10-09 by mcsauder