crayzeewulf/libserial #143: Build stuck on MacOSX

79rpm, 2019-06-23:
I am on macOS Sierra version 10.12.6. I issued the command .compile.sh in the root directory. I am getting an error related to a Linux header file. 

I thought the library was POSIX compatible, but I do not see a set of switches for the different OSes and I am running into this compilation issue. Please let me know if I am missing something,

Thank you,

xxx-MacBook-Pro:libserial-master xxx$ ./compile.sh 
+ mkdir -p build
+ cd build
+ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/xxx/dev/xxx/OSX/libserial-master/build
+ make
[ 12%] Building CXX object src/CMakeFiles/libserial_shared.dir/SerialPort.cpp.o
warning: unknown warning option '-Wchkp' [-Wunknown-warning-option]
warning: unknown warning option '-Wlogical-op'; did you mean '-Wlong-long'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-psabi' [-Wunknown-warning-option]
In file included from /Users/xxx/dev/xxx/OSX/libserial-master/src/SerialPort.cpp:34:
In file included from /Users/xxx/dev/xxx/OSX/libserial-master/src/libserial/SerialPort.h:36:
/Users/xxx/dev/xxx/OSX/libserial-master/src/libserial/SerialPortConstants.h:129:55: warning: use of the 'maybe_unused'
      attribute is a C++1z extension [-Wc++1z-extensions]
        explicit NotOpen(const std::string& whatArg [[maybe_unused]])
                                                      ^
/Users/xxx/dev/xxx/OSX/libserial-master/src/libserial/SerialPortConstants.h:144:59: warning: use of the 'maybe_unused'
      attribute is a C++1z extension [-Wc++1z-extensions]
        explicit AlreadyOpen(const std::string& whatArg [[maybe_unused]])
                                                          ^
/Users/xxx/dev/xxx/OSX/libserial-master/src/libserial/SerialPortConstants.h:159:58: warning: use of the 'maybe_unused'
      attribute is a C++1z extension [-Wc++1z-extensions]
        explicit OpenFailed(const std::string& whatArg [[maybe_unused]])
                                                         ^
/Users/xxx/dev/xxx/OSX/libserial-master/src/libserial/SerialPortConstants.h:176:59: warning: use of the 'maybe_unused'
      attribute is a C++1z extension [-Wc++1z-extensions]
        explicit ReadTimeout(const std::string& whatArg [[maybe_unused]])
                                                          ^
/Users/xxx/dev/xxx/OSX/libserial-master/src/SerialPort.cpp:40:10: fatal error: 'linux/serial.h' file not found
#include <linux/serial.h>
         ^~~~~~~~~~~~~~~~
7 warnings and 1 error generated.
make[2]: *** [src/CMakeFiles/libserial_shared.dir/SerialPort.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/libserial_shared.dir/all] Error 2
make: *** [all] Error 2

crayzeewulf, 2019-06-24:
@79rpm Sorry about that. LibSerial has definitely become linux-only recently and the description needs to be updated. 

It can probably be made to work on Mac OS X with some minor changes but I do not have access to a Mac OS X machine to test such changes. 

I will update the description.

mcsauder, 2019-06-24:
Hi @79rpm and @crayzeewulf .  It looks like the only item that `linux/serial.h` gets used for is `serial_struct serial_port_info {} ;` in the `GetAvailableSerialPorts()` method.  I've searched around a little bit for a MAC OS equivalent but haven't discovered a solution yet.

@79rpm , if you want, you can delete line 40 and lines 1886-1894 in SerialPort.cpp to get past the compiler error you described.  Doing so will likely render the `GetAvailableSerialPorts()` inoperable, but it would serve as a workaround.  If you happen to discover a solution to replacing `serial_struct` with a MAC OS equivalent we could also work on fixing the `GetAvailableSerialPorts()` method for posix compliance.  Let us know if you find a solution to that method implementation that works on MAC OS!

79rpm, 2019-06-25:
Thank you all for the inputs, appreciate! I am integrating this library into an emulation environment (OSX) as well as on an embedded platform (Linux). I will do the workaround so I can pass the compilation point on the emulation environment. I should have more time to look at a long term solution closer to the end of July.

crayzeewulf, 2019-06-25:
(Thanks for the hints, @mcsauder)

@79rpm Thanks for looking into it. Please let us know if you are able to build on OSX after the workaround suggested by @mcsauder .

mcsauder, 2019-09-11:
Hi @79rpm , does this issue still exist for you (with current master)?

79rpm, 2019-09-11:
I am not using the library, so it is not an issue for me at this time.

closed 2019-09-11 by 79rpm