crayzeewulf/libserial #149: New Debian Release

mcsauder, 2019-09-12:
Hi @crayzeewulf ,  I don't know how to do it... but we need a new release package... :)

crayzeewulf, 2019-09-23:
Mark, I think [CPack](https://cmake.org/cmake/help/v3.0/module/CPack.html) is the way to go. I am working on it.

mcsauder, 2019-09-25:
Hi @crayzeewulf , as I'm digging around to learn more on this topic I thought I'd post a link here to a new github feature that might be useful in the future as well: https://help.github.com/en/articles/about-github-package-registry#supported-clients-and-formats

LocutusOfBorg, 2020-02-17:
I'm doing it in the official archive, I didn't notice the move from sf to github.

mcsauder, 2020-02-23:
Hi @LocutusOfBorg , how can we assist?

crayzeewulf, 2020-04-04:
[This](https://github.com/crayzeewulf/libserial/tree/packaging_using_cpack) branch now contains Docker files that can build packages for RHEL/CentOS-7, Ubuntu-16.04, Ubuntu-18.04, and Buster. If these pass tests then we can merge into master. I am working on a script that will build packages for all supported distributions in one-shot using Docker. Coming soon (I hope).

mcsauder, 2020-05-26:
Libersial 1.0.0 is an Ubuntu 20.04 pacakge!  Congratulations @crayzeewulf !

closed 2020-05-26 by mcsauder

LocutusOfBorg, 2020-07-15:
:)

crayzeewulf, 2020-07-15:
> :)

Thanks for doing this! The CPack integration ought to help in the future.