crayzeewulf/libserial #150: The gap problem (64 bytes burst)

giancarlocp, 2019-09-29:
This may is beyond the lib scope, but let me ask.
I have noticed [a speed problem](https://github.com/giancarlocp/libserial/blob/master/examples/gap_report.ipynb).
The stream is done in 64 Bytes burst.

Why this behavior?

mcsauder, 2019-09-30:
Hi @giancarlocp , would you post the code you tested with? Thanks.

crayzeewulf, 2019-10-02:
@giancarlocp also I am unable to see the images in the Jupyter/iPython notebook you linked. Will it be possible for you to attach a PDF version to this issue?

crayzeewulf, 2019-10-02:
@giancarlocp Nevermind. I downloaded .ipynb file and generated a PDF copy. [Here](https://github.com/crayzeewulf/libserial/files/3679157/gap_report.pdf) it is.

crayzeewulf, 2019-10-02:
I am reading through [this](https://www.ftdichip.com/Documents/AppNotes/AN232B-04_DataLatencyFlow.pdf) document and Section 2.3 looks interesting:

> Data is received from USB to the PC by a polling method. The driver will request a certain amountof data from the USB scheduler.  This is done in multiples of 64 bytes.  The 'bulk packet size' onUSB is a maximum of 64 bytes.  

I wonder if this is happening in the hardware.

crayzeewulf, 2019-10-02:
Additional sources which appear to indicate that this is set in the FTDI hardware:

[Link](https://analoghome.blogspot.com/2017/08/ftdi-ft232r-usb-to-serial-bridge.html)
> Packet Size – In these kinds of USB transfers the basic data packet size is 64 bytes. FTDI reserves 2 byes for it’s own use, so the user gets 62 bytes of data to use in every USB packet transfer. The packet size is fixed by the way USB works and can’t be changed.

[Link](https://www.ftdichip.com/Support/Documents/TechnicalNotes/TN_103_FTDI_USB_Data_Transfer_Efficiency(FT_000097).pdf)
> The result of this is that the data will be written to the device with 64 bytes per USB packet.  

[Link](http://www.bidib.org/support/serial_rx_e.html)
> For an example, we have a closer look to the popular FTDI FT232RL:
> The USB side acts with a packet size of 64 bytes, which is in detail 2 bytes of status and 62 bytes for user data.

[Link](https://www.ftdichip.com/Support/Knowledgebase/index.html?an232b_03optimizingperformance.htm)
> The question of how to setup an application for optimized data throughput is complicated by the format of the data that is transferred over USB. Data is transferred over USB in a series of 64 byte packets. Each packets contains two bytes that are reserved by FTDI and 62 bytes of actual data. So a 4Kbyte USB transaction contains 3968 bytes of actual data, and a 64Kbyte transaction contains 63488 bytes. In general, an x byte USB transaction yields x - ((x/64)*2) bytes of actual data.

giancarlocp, 2019-10-15:
The problem is just the **CH340G doesn't have double buffer**.
It has to clean and fill the buffer before start a new **64B burst** transmission.
I have tested the same code with the **CP2101**
and it works with higher rate (near **99kB/s**) at **1.5MHz** (its max baud rate).

crayzeewulf, 2019-10-15:
@giancarlocp Thanks for confirming this. Is it okay to close this issue then?

closed 2019-10-15 by giancarlocp