crayzeewulf/libserial #151: Allow move construction

raymoo, 2019-10-08:
Is there anything preventing `SerialStream` from being move constructible? It has prevented us from factoring our stream creation into a function that returns a `SerialStream` without pointer indirection.

crayzeewulf, 2019-10-09:
`SerialStream` only contains a `std::unique_ptr<SerialStreamBuf>` as a data member. So, in principle, a move constructor can be written by simply moving this data member (or swapping it). Will it be possible for you to try it out in your setup? Something like the following should be sufficient:

```cpp
SerialStream::SerialStream(const SerialStream&& otherSerialStream) : 
    mIOBuffer(std::move(otherSerialStream.mIOBuffer))
{
    // empty
}
```

It might also be useful to enable move assignment if move construction is allowed.

raymoo, 2019-10-09:
Couldn't I just do `SerialStream::SerialStream(SerialStream&& otherSerialStream) = default`? (move constructor takes a not-const rvalue reference because the move can modify the object).

I don't have the opportunity to touch the code immediately, but I can try it when I do.

crayzeewulf, 2019-10-10:
@raymoo you are right on both counts: move constructor arguments must be non-const rvalue reference and `default` is the right way to do this. I blame sleep deprivation. :D

In any case, if you do try this out, please let us know so we can add it to the library code.

mcsauder, 2019-10-10:
Hi @crayzeewulf and @raymoo .  I thought I'd give this a try, but something is eluding me.. My addition to the unit tests for checking copy/move assignment is not compiling.  Let me know if you can spot what I don't understand!  https://github.com/crayzeewulf/libserial/tree/copy_and_move_constructors

crayzeewulf, 2019-10-10:
Mark:

I have not tried this but the `default` move constructor for `SerialStream` should work without any error when placed in the header file `SerialStream.h`. However, it will (probably) not work for `SerialStreamBuf` and `SerialPort` because the definition of their corresponding `Implementation` classes are (intentionally) hidden in the `.cpp` files. 

The workaround for this is to declare the move constructor in the header and define it in the `.cpp` file. For example, in `SerialPort.h`:

```cpp
class SerialPort
{
public:
    // ...
    SerialPort(SerialPort&& otherSerialPort) ;
    // ...
} ;
```

and then put the definition in `SerialPort.cpp`:

```cpp
SerialPort::SerialPort(SerialPort&& otherSerialPort) = default ;
```

Please give this a try and let me know what happens. If you still get an error, could you please also post the error message (I am not going to be at a computer where I can try this out till Friday).

Thanks as always!

crayzeewulf, 2019-10-10:
P.S. Also, the other three (copy constructor, copy assignment, and move assignment) should be left `delete`d. Eventually, it might be useful to enable move assignment as well. But till then we want to leave it deleted (following [rule of five](https://en.cppreference.com/w/cpp/language/rule_of_three)). We probably do not want to support copy constructor and copy assignment as the underlying file descriptor may not be be usable from two different instance of these classes.

crayzeewulf, 2019-10-10:
PPS. (as the car-talk guys would say): More about this at [C++ Core Guidelines](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#Rc-five).