crayzeewulf/libserial #153: have problems in using libserial

CrazyPopLin, 2019-10-30:
Hi,
I am using ubuntu 16.04 and installed libserial by 

> sudo apt install libserial-dev 

And I looked at the **example_project**, when I build this project,  **cmake** operation is fine. But when I use **make** operation, I have error:

> fatal error: libserial/SerialPort.h: No such file or directory  

I read the NOTE in CMakeLists.txt, thought the libserial installed by apt-get should be fine. I just found **SerialStreamBuf.h** , **SerialStreamBuf.h** and **SerialStream.h** in **/usr/include**.

Could you give me any suggestions to use libserial on my project in CMakeLists.txt? Thanks for your help.

mcsauder, 2019-10-31:
Hi @CrazyPopLin , apologies you ran into issues with the debian package, (#149).  Try removing libserial using apt, then install the current library using the instructions in the [README](https://github.com/crayzeewulf/libserial/blob/master/README.md)

Apologies for the inconvenience, we still need to get a new release out, it's been too long!

-Mark

closed 2019-10-31 by CrazyPopLin

mcsauder, 2019-10-31:
Let us know if you have any troubles getting things to work for you!