crayzeewulf/libserial #155: Using a Constructor to copy objects

chaz765, 2019-11-26:
Hi,  
Probably a design question. 
I tried to instantiate a constructor with the SerialPort object and found that it is not allowed by the library. Was wondering why that was done ? And what would be the right way to instantiate an object using a constructor ?
Would commenting out the part of your code which blocks this behavior cause any problems ?

crayzeewulf, 2019-11-27:
@chaz765, I am guessing that you are talking about the [copy constructor](https://en.cppreference.com/w/cpp/language/copy_constructor) for `SerialPort` class and not just a constructor (as multiple constructors are available to create instances of `SerialPort` class: see [this](https://github.com/crayzeewulf/libserial/blob/v1.0.0/src/libserial/SerialPort.h#L76) and [this](https://github.com/crayzeewulf/libserial/blob/v1.0.0/src/libserial/SerialPort.h#L62)).

Copy construction and copy assignment are deliberately disabled because creating a copy of `SerialPort` would imply that multiple instances would share the same file descriptor and try to access the same serial port device. This could be problematic. Modifying properties of one copy will affect the other copies implicitly. This would be non-intuitive from a user's perspective. Similarly, when one copy goes out of scope (or is otherwise destroyed), it might close the underlying file descriptor thus preventing all other copies to access the serial port subsequently. There are other similar issues that drive the need to disallow copy construction and assignment. 

Move construction and assignment should be safe and are planned to be implemented shortly (see issue #151). 

Could you please post some sample code showing how you are attempting to use `SerialPort` and why you need a copy constructor? There might be a simple way around it. For example, you might be able to create a [std::shared_ptr<SerialPort>](https://en.cppreference.com/w/cpp/memory/shared_ptr) instead and make as many copies of it as you want.

crayzeewulf, 2020-06-07:
@chaz765 Do you have any additional information regarding this? If not, can we close this issue?

cnanda-zbr, 2020-06-08:
Yes please thank you. I was able to use the code in a way that worked eventually. I did not use a copy constructor.

crayzeewulf, 2020-06-08:
Thanks for the update.

closed 2020-06-08 by crayzeewulf