crayzeewulf/libserial #156: error: declaration of ‘use_UnorderedPointwise_with_hash_tables’ shadows a member of 'this'

cpaolini, 2019-12-21:
Build error on CentOS release 6.10 (Final):

error: declaration of ‘use_UnorderedPointwise_with_hash_tables’ shadows a member of 'this'

-- Build files have been written to: /home/paolini/libserial/build/gtest/src/GTestExternal-build
[ 15%] Performing build step for 'GTestExternal'
Scanning dependencies of target gtest
[ 12%] Building CXX object googlemock/gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 25%] Linking CXX static library libgtest.a
[ 25%] Built target gtest
Scanning dependencies of target gmock
[ 37%] Building CXX object googlemock/CMakeFiles/gmock.dir/src/gmock-all.cc.o
cc1plus: warnings being treated as errors
In file included from /home/paolini/libserial/build/gtest/src/GTestExternal/googlemock/include/gmock/gmock-spec-builders.h:71,
                 from /home/paolini/libserial/build/gtest/src/GTestExternal/googlemock/include/gmock/gmock-generated-function-mockers.h:44,
                 from /home/paolini/libserial/build/gtest/src/GTestExternal/googlemock/include/gmock/gmock.h:62,
                 from /home/paolini/libserial/build/gtest/src/GTestExternal/googlemock/src/gmock-all.cc:39:
/home/paolini/libserial/build/gtest/src/GTestExternal/googlemock/include/gmock/gmock-matchers.h: In member function ‘testing::internal::PointwiseMatcher<TupleMatcher, RhsContainer>::operator testing::Matcher<T>() const’:
/home/paolini/libserial/build/gtest/src/GTestExternal/googlemock/include/gmock/gmock-matchers.h:3028: error: declaration of ‘use_UnorderedPointwise_with_hash_tables’ shadows a member of 'this'
make[5]: *** [googlemock/CMakeFiles/gmock.dir/src/gmock-all.cc.o] Error 1
make[4]: *** [googlemock/CMakeFiles/gmock.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [gtest/src/GTestExternal-stamp/GTestExternal-build] Error 2
make[1]: *** [CMakeFiles/GTestExternal.dir/all] Error 2
make: *** [all] Error 2
[paolini@host libserial]$

mcsauder, 2019-12-21:
@cpaolini , it looks like this is an issue with GTest rather than libserial...  I cannot reproduce the error on Ubuntu 19.10 with GTest 1.8.1 but your reporting did alert me to the fact that Gtest has jumped from 1.8.1 to 1.10.0, and v1.10 is not immediately compatible.

I will put some work into updating the unit tests to use GTest 1.10.0, but it might take some time.  I'll report back as I make progress.

In the meantime, here is a WIP branch you are free to work from as well: https://github.com/crayzeewulf/libserial/tree/gtest_update

I'll keep everyone posted with any updates.

-Mark

cpaolini, 2019-12-21:
Thank you.  I switched to the gtest_update branch, but am now having a missing declaration issue
with nullptr in gtest-port.h: 

[paolini@kon libserial]$ ./compile.sh
+ mkdir -p build
+ cd build
+ cmake3 ..
-- Boost version: 1.41.0
-- Found the following Boost libraries:
--   unit_test_framework
-- Configuring done
-- Generating done
-- Build files have been written to: /home/paolini/libserial/build
+ make
[  2%] Performing build step for 'GTestExternal'
[ 12%] Building CXX object googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
In file included from /home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-internal.h:40,
                 from /home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/gtest.h:62,
                 from /home/paolini/libserial/build/gtest/src/GTestExternal/googletest/src/gtest-all.cc:38:
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h: In function ‘void testing::internal::FlushInfoLog()’:
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h:972: error: ‘nullptr’ was not declared in this scope
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h: In function ‘To testing::internal::DownCast_(From*)’:
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h:1082: error: ‘nullptr’ was not declared in this scope
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h:1088: error: ‘nullptr’ was not declared in this scope
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h: In function ‘void testing::internal::SleepMilliseconds(int)’:
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h:1157: error: ‘nullptr’ was not declared in this scope
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h: In constructor ‘testing::internal::Notification::Notification()’:
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h:1175: error: ‘nullptr’ was not declared in this scope
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h: In function ‘void* testing::internal::ThreadFuncWithCLinkage(void*)’:
/home/paolini/libserial/build/gtest/src/GTestExternal/googletest/include/gtest/internal/gtest-port.h:1285: error: ‘nullptr’ was not declared in this scope

mcsauder, 2026-10-07:
I believe this was fixed with the addition of gtest 1.17 and C++17 in commits 50e0f44 and c1caa91.  Please reopen if this issue remains.

closed 2026-10-07 by mcsauder