crayzeewulf/libserial #159: Trying to get the example_project (Noob perspective)

alirezasafadri, 2020-04-04:
I followed the following steps, but I cannot get the sample project to compile.
 
> If you simply want to use LibSerial and you already utilize a Debian Linux distribution, use apt to install the current release package:
sudo apt install libserial-dev
Example code to demonstrate how to use the library can be found in the examples directory. An self-contained example project demonstrating the use of CMake and GNU Autotools (make) can be found in examples/example_project directory.

My cmake version is 3.17.0. I get the following error when
> sudo ./compile.sh

I get the following when I run it:
> -- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/Lora/stage3/build
make: Entering directory '/home/pi/Desktop/Lora/stage3/build'
make[1]: Entering directory '/home/pi/Desktop/Lora/stage3/build'
make[2]: Entering directory '/home/pi/Desktop/Lora/stage3/build'
Scanning dependencies of target ExampleProject
make[2]: Leaving directory '/home/pi/Desktop/Lora/stage3/build'
make[2]: Entering directory '/home/pi/Desktop/Lora/stage3/build'
[ 50%] Building CXX object CMakeFiles/ExampleProject.dir/example_project.cpp.o
/home/pi/Desktop/Lora/stage3/example_project.cpp:5:10: fatal error: libserial/SerialPort.h: No such file or directory
 #include <libserial/SerialPort.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ExampleProject.dir/build.make:80: CMakeFiles/ExampleProject.dir/example_project.cpp.o] Error 1
make[2]: Leaving directory '/home/pi/Desktop/Lora/stage3/build'
make[1]: *** [CMakeFiles/Makefile2:93: CMakeFiles/ExampleProject.dir/all] Error 2
make[1]: Leaving directory '/home/pi/Desktop/Lora/stage3/build'
make: *** [Makefile:101: all] Error 2
make: Leaving directory '/home/pi/Desktop/Lora/stage3/build'

I am a noob so please accept my apologies if I am missing something obvious.

crayzeewulf, 2020-04-04:
@alirezasafadri I am sorry about that. 

If you installed the `libserial-dev` package, it is likely that you have an older version of the library. In order to resolve this, please do the following:

1. Remove `libserial-dev` package.
1. Compile and install LibSerial from the latest source code in the `master` branch on GitHub as described [here](https://github.com/crayzeewulf/libserial#developers).
1. Try the example project again.

If you are not comfortable compiling LibSerial from source code, please let us know which Linux distribution you are using and we will provide a binary package for you that might work.

alirezasafadri, 2020-04-04:
@crayzeewulf Please do not say that you are sorry. You are awesome and I highly appreciate everything you have done for this answer and this repository.

I removed the `libserial-dev` and tried the developer directions and used  cmake but when i do `sudo ./compile.sh` I get the following error:
```
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'libserial'
--   No package 'libserial' found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/Lora/stage4/build
make: Entering directory '/home/pi/Desktop/Lora/stage4/build'
make[1]: Entering directory '/home/pi/Desktop/Lora/stage4/build'
make[2]: Entering directory '/home/pi/Desktop/Lora/stage4/build'
Scanning dependencies of target ExampleProject
make[2]: Leaving directory '/home/pi/Desktop/Lora/stage4/build'
make[2]: Entering directory '/home/pi/Desktop/Lora/stage4/build'
[ 50%] Building CXX object CMakeFiles/ExampleProject.dir/example_project.cpp.o
[100%] Linking CXX executable ExampleProject
/usr/bin/ld: CMakeFiles/ExampleProject.dir/example_project.cpp.o: in function `main':
example_project.cpp:(.text+0x14): undefined reference to `LibSerial::SerialPort::SerialPort()'
/usr/bin/ld: example_project.cpp:(.text+0x20): undefined reference to `LibSerial::SerialStream::SerialStream()'
/usr/bin/ld: example_project.cpp:(.text+0x68): undefined reference to `LibSerial::SerialPort::Open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::_Ios_Openmode const&)'
/usr/bin/ld: example_project.cpp:(.text+0xc8): undefined reference to `LibSerial::SerialStream::Open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::_Ios_Openmode const&)'
/usr/bin/ld: example_project.cpp:(.text+0xfc): undefined reference to `LibSerial::SerialPort::SetBaudRate(LibSerial::BaudRate const&)'
/usr/bin/ld: example_project.cpp:(.text+0x118): undefined reference to `LibSerial::SerialStream::SetBaudRate(LibSerial::BaudRate const&)'
/usr/bin/ld: example_project.cpp:(.text+0x14c): undefined reference to `LibSerial::SerialPort::WriteByte(char)'
/usr/bin/ld: example_project.cpp:(.text+0x180): undefined reference to `LibSerial::SerialPort::ReadByte(char&, unsigned int)'
/usr/bin/ld: example_project.cpp:(.text+0x200): undefined reference to `LibSerial::SerialPort::Close()'
/usr/bin/ld: example_project.cpp:(.text+0x20c): undefined reference to `LibSerial::SerialStream::Close()'
/usr/bin/ld: example_project.cpp:(.text+0x218): undefined reference to `LibSerial::SerialStream::~SerialStream()'
/usr/bin/ld: example_project.cpp:(.text+0x224): undefined reference to `LibSerial::SerialPort::~SerialPort()'
/usr/bin/ld: example_project.cpp:(.text+0x2c8): undefined reference to `LibSerial::SerialStream::~SerialStream()'
/usr/bin/ld: example_project.cpp:(.text+0x2d8): undefined reference to `LibSerial::SerialPort::~SerialPort()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ExampleProject.dir/build.make:101: ExampleProject] Error 1
make[2]: Leaving directory '/home/pi/Desktop/Lora/stage4/build'
make[1]: *** [CMakeFiles/Makefile2:93: CMakeFiles/ExampleProject.dir/all] Error 2
make[1]: Leaving directory '/home/pi/Desktop/Lora/stage4/build'
make: *** [Makefile:101: all] Error 2
make: Leaving directory '/home/pi/Desktop/Lora/stage4/build'
```
I do not know how to reverse the developer directions, but if you think it would be necessary, I will do the installation again.

I am using Raspbian on Raspberry Pi 4, to be honest I do not want to be that much of burden for you and I am noob enough not to know how difficult that would be :) but it sounds difficult. So please try to put  food on my plate rather than spending your time for this. If you think I should read on any topic or link, I would be more than happy to that.

I am sorry for being very unaware of these stuff. I am very new to Linux.

crayzeewulf, 2020-04-04:
@alirezasafadri Thanks for your feedback. I had not used the example project in a while and just noticed that the `CMakeLists.txt` file in it says the following:

```
# **NOTE** This example currently requires the LibSerial be built using
# autotools as the CMake build of LibSerial does not install a pkg-config
# file yet. Support for pkg-config will be added to CMake build shortly.
```

Could you please rebuild LibSerial using `autotools` instead of `CMake` and let us know if that works? We will work on fixing these issues so the example project works either way.

alirezasafadri, 2020-04-04:
@crayzeewulf sorry for not understanding that part. I am very new to these words so sometimes they do not click the right way.
Thank you and I will test it shortly and let you know if I got it working.

alirezasafadri, 2020-04-04:
I removed the files from `/usr/local/lib`

I followed the steps in `autotools` sections.

The example projects is built without any issue:
`-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'libserial'
--   Found libserial, version 1.0.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/InstalledLibraries/libserial-master/examples/example_project/build
make: Entering directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
make[1]: Entering directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
make[2]: Entering directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
Scanning dependencies of target ExampleProject
make[2]: Leaving directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
make[2]: Entering directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
[ 50%] Building CXX object CMakeFiles/ExampleProject.dir/example_project.cpp.o
[100%] Linking CXX executable ExampleProject
make[2]: Leaving directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
[100%] Built target ExampleProject
make[1]: Leaving directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
make: Leaving directory '/home/pi/InstalledLibraries/libserial-master/examples/example_project/build'
`

However when I `./ExampleProject`, I get the following error:
`./ExampleProject: error while loading shared libraries: libserial.so.1: cannot open shared object file: No such file or directory`

I can see `libserial.so.1` in `/usr/local/lib`.

crayzeewulf, 2020-04-04:
@alirezasafadri You are almost there!

The first thing to try is to run `ldconfig` and then try running the example again:

```sh
sudo ldconfig
```

If this does not work then the folder `/usr/local/lib` is probably not being searched for libraries by your dynamic linker/loader. You can fix this by setting the `LD_LIBRARY_PATH` variable using the following command before running the example program:

```sh
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
```

Let us know if either one of those work.

alirezasafadri, 2020-04-04:
I tried `sudo ldconfig` and ran the code and it worked.

Thank you very much. Is there anyway I can help back?

crayzeewulf, 2020-04-04:
> 
> I tried `sudo ldconfig` and ran the code and it worked.
>

Awesome! Glad that worked.
 
> Thank you very much. Is there anyway I can help back?

Thanks and you've helped already by providing feedback and testing the library. Using the library and letting us know if parts of it do not work will be even more helpful. It will help us fix bugs and make it better. 

If you are satisfied with the solution to this issue, please close this ticket.

closed 2020-04-04 by alirezasafadri