crayzeewulf/libserial #162: SerialStream eats up feed characters 0x0A, 0x0D, and 0x0F when reading

JacekRuzyczka, 2020-06-07:
When communicating with a serial-over-USB device (a GNS FM9 receiver for traffic messages) using a `SerialStream` object, I found out that not all characters coming from the device are in fact being read. This is what I always get served:

`3F 00 56 08 37 3F`

When using the same device with the [RDS Surveyor](https://github.com/ChristopheJacquet/RdsSurveyor) app (which is written in Java and uses a `com.fazecast.jSerialComm.SerialPort)`, the response looks like this:

`3F 00 00 56 08 0A 37 0C 0D 3F`

After some research I stumbled over this posting: [Wrong newline character over serial port (CR instead of LF)](https://superuser.com/questions/714078/wrong-newline-character-over-serial-port-cr-instead-of-lf). The poster does not use `libserial`, but his issue is exactly the same as mine.

This is the system-side config of the port in question:

`pi@autoradio:~ $ stty -a
speed 38400 baud; rows 40; columns 80; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>;
eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R;
werase = ^W; lnext = ^V; discard = ^O; min = 1; time = 0;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread -clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr icrnl ixon ixoff
-iuclc -ixany -imaxbel iutf8
opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
isig icanon iexten echo echoe echok -echonl -noflsh -xcase -tostop -echoprt
echoctl echoke -flusho -extproc`

The system is a Raspbery Pi 3B with the latest production version of Raspbian running.

Now, my question is: How can I make my `SerialStream` read all characters without any transformations, just as they come from the port? I don't really wanna use `stty` because its settings might get overwritten when removing and reattaching the device. Thank you.

**NOTE:** If necessary, I can post the code in question, but it's quite long.

crayzeewulf, 2020-06-07:
@JacekRuzyczka Some sample code will surely help but I am guessing that you may be attempting to read data using `>>` operator. This operator is mostly used for [formatted input](https://en.cppreference.com/w/cpp/named_req/FormattedInputFunction) and skips whitespaces by default. You can use [std::noskipws](https://en.cppreference.com/w/cpp/io/manip/skipws) to help with that. For binary data, it might be better to use [unformatted input](https://en.cppreference.com/w/cpp/named_req/UnformattedInputFunction) methods as described under the "Unformatted Input" section [here](https://en.cppreference.com/w/cpp/io/basic_istream). Alternatively, you may consider using the `SerialPort` class instead of `SerialStream` as it will provide you with raw data without any translation or skipping.

JacekRuzyczka, 2020-06-07:
@crayzeewulf I'm already using the operator, just like this:
```
SerialStream tmc_receiver_handler ("/dev/ttyUSB0", ios::in | ios::out);
char c = 0x00;

tmc_receiver_handler.SetBaudRate      (SerialStreamBuf::BAUD_38400);
tmc_receiver_handler.SetCharSize      (SerialStreamBuf::CHAR_SIZE_8);
tmc_receiver_handler.SetFlowControl   (SerialStreamBuf::FLOW_CONTROL_NONE);
tmc_receiver_handler.SetParity        (SerialStreamBuf::PARITY_NONE);
tmc_receiver_handler.SetNumOfStopBits (SerialPort::STOP_BITS_1);

tmc_receiver_handler >> c;
```

crayzeewulf, 2020-06-07:
@JacekRuzyczka The use of the `>>` operator is likely the cause of the problem. Please try using one of the following and let us know if it works. 

1. Change the last line to:
```cpp
tmc_receiver_handler >> std::noskipws >> c;
```
2. Or, try replacing it with an unformatted input function such as:
```cpp
c = tmc_receiver_handler.get() ;
```

JacekRuzyczka, 2020-06-08:
Your proposal worked immediately. Thank you!

`skipws` is a feature of the C++ `std::iostream`, right?

crayzeewulf, 2020-06-08:
Great to hear that! Thanks for letting us know. 

And, yes, `skipws` is part of the C++ standard input/output library.

closed 2020-06-08 by crayzeewulf