crayzeewulf/libserial #169: Compile error Cmake

Pantau123, 2020-09-21:
Hi i want to build libserial, but i receive the following error:
[1/5] Building CXX object _deps/libserial-build/src/CMakeFiles/libserial_shared.dir/SerialPort.cpp.o
FAILED: _deps/libserial-build/src/CMakeFiles/libserial_shared.dir/SerialPort.cpp.o 
/opt/karo-minimal/1.0/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -march=armv7ve -mthumb -mfpu=neon -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/karo-minimal/1.0/sysroots/armv7vet2hf-neon-poky-linux-gnueabi --sysroot=/opt/karo-minimal/1.0/sysroots/armv7vet2hf-neon-poky-linux-gnueabi -Dlibserial_shared_EXPORTS -I_deps/libserial-build/gtest/src/GTestExternal/googletest/include -I_deps/libserial-src/src -O2 -pipe -g -feliminate-unused-debug-types  -g -fPIC   -Wall -Wcast-align -Wchar-subscripts -Wchkp -Wdouble-promotion -Wextra -Wfatal-errors -Wformat -Wformat-security -Wlogical-op -Wno-format-extra-args -Wno-long-long -Wno-parentheses -Wno-psabi -Wno-variadic-macros -Woverlength-strings -Wpacked -Wpointer-arith -Wunused-local-typedefs -Wwrite-strings -fstrict-aliasing -fno-check-new -fno-common -fvisibility=default -pedantic -pthread -std=gnu++14 -MD -MT _deps/libserial-build/src/CMakeFiles/libserial_shared.dir/SerialPort.cpp.o -MF _deps/libserial-build/src/CMakeFiles/libserial_shared.dir/SerialPort.cpp.o.d -o _deps/libserial-build/src/CMakeFiles/libserial_shared.dir/SerialPort.cpp.o -c _deps/libserial-src/src/SerialPort.cpp
arm-poky-linux-gnueabi-g++: warning: switch '-Wchkp' is no longer supported
_deps/libserial-src/src/SerialPort.cpp: In member function 'void LibSerial::SerialPort::Implementation::Read(LibSerial::DataBuffer&, size_t, size_t)':
_deps/libserial-src/src/SerialPort.cpp:2211:71: error: no matching function for call to 'max(const size_t&, long unsigned int)'
 2211 |         size_t number_of_bytes_remaining = std::max(numberOfBytes, 1UL) ;
      |                                                                       ^
compilation terminated due to -Wfatal-errors.

mcsauder, 2020-09-23:
Hi @Pantau123 , 

Could you check out [this branch](https://github.com/crayzeewulf/libserial/tree/static_casts) and report back if it succeeds for you?  I think that the current master branch would succeed for you now as-is, but using a static cast here is probably preferable anyway. If it succeeds we'll get that merged in.

Thanks for your feedback!

-Mark

Pantau123, 2020-09-29:
I have tested with the main branch and this works...

closed 2020-09-29 by Pantau123

crayzeewulf, 2020-09-29:
> I have tested with the main branch and this works...

Thanks for the update and testing this.