crayzeewulf/libserial #170: Problem with receiving SerialStream

Pantau123, 2020-09-29:
Hi, i use the main branch version of Seriallib with SerialStream.
I can not  read characters with << from the SerialStream.
I can only read the function SerialStream.read(char*,streamsize), but everytime i receive a leading '\0' and i have to create a char array with receivedbytes +1.
Can anybody help me?
Bests 
Pantau
File main.cpp (sender)
`#include <cstdio>

#include <libserial/SerialStream.h>

using namespace LibSerial;
using namespace std;

int main()
{
	SerialStream m_serialPort;
	BaudRate defaultBaudRate = BaudRate::BAUD_9600;
	CharacterSize defaultCharacterSize = CharacterSize::CHAR_SIZE_8;
	FlowControl defaultFlowControl = FlowControl::FLOW_CONTROL_NONE;
	Parity defaultParity = Parity::PARITY_NONE;
	StopBits defaultStopBits = StopBits::STOP_BITS_1;
	int defaultBufferSize = 50;
	m_serialPort.SetBaudRate(defaultBaudRate);
	m_serialPort.SetCharacterSize(defaultCharacterSize);
	m_serialPort.SetFlowControl(defaultFlowControl);
	m_serialPort.SetParity(defaultParity);
	m_serialPort.SetStopBits(defaultStopBits);
	m_serialPort.SetVMin(0);
	m_serialPort.SetVTime(0);
	m_serialPort.FlushIOBuffers();
	m_serialPort.Open("/dev/ttyS1");
	std::string my_string = "Hello, Serial Port.";
	m_serialPort << my_string << endl;
	m_serialPort.DrainWriteBuffer();
	m_serialPort.Close();
    return 0;
}`
File main.cpp (receiver)
`#include <cstdio>
#include <libserial/SerialStream.h>
#include <sstream>
#include <unistd.h>
#include <iostream>
using namespace LibSerial;
using namespace std;
int main()
{
    SerialStream m_serialPort;
	BaudRate defaultBaudRate = BaudRate::BAUD_9600;
	CharacterSize defaultCharacterSize = CharacterSize::CHAR_SIZE_8;
	FlowControl defaultFlowControl = FlowControl::FLOW_CONTROL_NONE;
	Parity defaultParity = Parity::PARITY_NONE;
	StopBits defaultStopBits = StopBits::STOP_BITS_1;
	int defaultBufferSize = 50;
	m_serialPort.SetBaudRate(defaultBaudRate);
	m_serialPort.SetCharacterSize(defaultCharacterSize);
	m_serialPort.SetFlowControl(defaultFlowControl);
	m_serialPort.SetParity(defaultParity);
	m_serialPort.SetStopBits(defaultStopBits);
	m_serialPort.SetVMin(0);
	m_serialPort.SetVTime(0);

	
	
	auto serialPorts = m_serialPort.GetAvailableSerialPorts();
	m_serialPort.Open("/dev/ttyS2");
	m_serialPort.FlushIOBuffers();
	stringstream rcvBuffer;
	
	bool foundString = false;

	while (foundString!=true)
	{
		bool streamGood = false;
		while (m_serialPort.rdbuf()->in_avail() == 0)
		{
			usleep(50);
		}
		int receivingBytes = m_serialPort.GetNumberOfBytesAvailable();
		int read = m_serialPort.rdbuf()->in_avail();
		char character[read+1];//+1 because have a trailling zero
		m_serialPort.read(character, read+1);
		cout << character << endl;
		
	}
	m_serialPort.Close();
    return 0;
}`

crayzeewulf, 2020-09-29:
Are you expecting null bytes or whitespaces in your input. Maybe have a look at #162.

Pantau123, 2020-09-29:
No i don't expect a null byte but it can be a whitespace.

Pantau123, 2020-09-29:
But the problem is where does the '\0' come from?

Pantau123, 2020-09-29:
with noskipws i receive the data with <<

mcsauder, 2021-03-25:
Hi @Pantau123 , did that resolve this issue for you?

mcsauder, 2025-09-15:
Closing as stale, hopefully this was resolved!

closed 2025-09-15 by mcsauder