crayzeewulf/libserial #173: Input/output error

mrspirytus, 2021-03-19:
Hello,

I am getting exceptions occasionally on Write(). Here is an exception I am getting.

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Input/output error
Aborted (core dumped)
```

here is kernel dmsg. Did you ever encounter this? I have a little LTE module plugged into 2 port USB hub. And sending AT commands on  /dev/ttyUSB1. I am not sure if this is a problem with liberalism. I am still investigating 
```
disabled by hub (EMI?), re-enabling...
```

Cheers

mcsauder, 2021-03-25:
Hi @mrspirytus , are you using a SerialPort or a SerialStream object?

mcsauder, 2021-03-25:
You will get this exception in the following case:
```
else if (write_result <= 0 && errno != EWOULDBLOCK)
{
        throw std::runtime_error(std::strerror(errno)) ;
}
```
So if the write was unsuccessful and reported back as such by the OS, or if errno is anything but a blocking call you will see this.  If you add a try/catch block to your executing code you can capture this exception and continue execution of your application.

mrspirytus, 2021-04-15:
Still looking into this...

mcsauder, 2026-10-07:
Hi @mrspirytus , I dug into this a little with Claude.  The kernel log shows the USB hub disabling the port ("disabled by hub (EMI?)"). indicating a hardware or USB problem, and that libserial is correctly reporting the error.

I'd like to close this issue, but we can re-open it if you still think it's a libserial error that we can track down.

closed 2026-10-07 by mcsauder