crayzeewulf/libserial #174: DrainIOBuffers not in libserial-dev

sousarbarb, 2021-03-25:
Hi,

I installed the LibSerial library through the `sudo apt install libserial-dev`.

However, when implementing communication through serial ports using threads, as already noticed in the issues, I had several problems.

I've seen in https://github.com/crayzeewulf/libserial/issues/99 that the call of DrainIOBuffer resolves the problem for sleep. However, this member function is not implemented in libserial-dev.

Could the package be updated?

mcsauder, 2021-08-05:
Hi @sousarbarb , are you running on a recent debian distro?  The latest release only made it into Ubuntu 20.04.  Can you provide details on your OS?  Thanks for submitting your issue.

mcsauder, 2025-09-15:
Hi @sousarbarb , hopefully this was resolved for you.  Let us know if not and I can re-open.