crayzeewulf/libserial #175: Linking using example does not work

edwardribbit, 2021-04-27:
Hi,

I was having issues with linking with libserial.

What I did:
System: Ubuntu 20
`sudo apt install libserial-dev`
Followed the example CMakeLists for linking
`target_link_libraries(ExampleProject PRIVATE ${SERIAL_LDFLAGS} ${CMAKE_THREAD_LIBS_INIT})`

What I saw:
Symbol lookup error

It worked after I changed `SERIAL_LDFLAGS` to `SERIAL_LIBRARIES`
Perhaps an update of the example is required?

mcsauder, 2021-04-29:
Hi @edwardribbit , I just purged all libserial files from the /usr/* directories on my Ubuntu 20.10 machine, then compiled the example project using the compile script and observed it failing to compile for lack of included headers.  I then reinstalled libserial-dev using apt and everything compiles fine without modifications when I run the compile script.

I am also able to compile successfully with the modification you proposed above.

Could you try purging libserial, `sudo apt purge libserial-dev`, then update your packages, `sudo apt update`, then reinstall and see if you are still seeing the same results reported above?

edwardribbit, 2021-04-29:
I forgot to mention i'm in a docker container.

After deleting and rebuilding the container, both methods of linking worked.
From my understanding there isn't a standard syntax to CMake variables, is SERIAL_LDFLAGS or SERIAL_LIBRARIES preferred and why?

This issue can be closed.

mcsauder, 2021-05-04:
Thanks for reporting back!

mcsauder, 2021-05-04:
>From my understanding there isn't a standard syntax to CMake variables, is SERIAL_LDFLAGS or SERIAL_LIBRARIES preferred and why?

@crayzeewulf might know, I really don't have an answer.

closed 2021-05-04 by mcsauder

crayzeewulf, 2021-05-04:
> 
> @crayzeewulf might know, I really don't have an answer.

Hmm. I don't recall offhand. Let me check [Professional CMake](https://crascit.com/professional-cmake/) to see what is recommended.