crayzeewulf/libserial #177: Writing blocks of data

dollop80, 2021-08-04:
Hi,
The documentation states that there is a possibility to send a data buffer with a predefined length https://libserial.readthedocs.io/en/latest/tutorial.html#writing-blocks-of-data
But seems that this is not implemented because there no function Write, that conforms Write( output_buffer, BUFFER_SIZE );
So is there a way to switch the library to binary mode anyway? To send, f.e. a 0x00 byte?

mcsauder, 2021-08-05:
Hi @dollop80 , nice catch!  It indeed looks like this can only be accomplished through a SerialStream object at present.  Would you be interested in submitting a PR?

Thanks for submitting the issue!

mcsauder, 2026-10-07:
Resolved with PR #202