crayzeewulf/libserial #180: Compile error in 1.10.0 while using GCC 11

IcenDy, 2021-12-03:
This error may be caused by googletest 1.8.1, but there is also someting error when I modify version and usr_hash in CmakeLists.txt.

mcsauder, 2021-12-07:
Hi @IcenDy , 

Yes, this issue is due to GTest 1.8.1 vs 1.8.0, I am embarrassed to admit that I recognize this and have not fixed it. (However, shame on the GTest devs for not following Semantic versioning, https://semver.org/... Seriously, shame on them.)

If you are able to resolve the update to 1.8.1, please submit a PR, and in the meantime you might have spurred me on to working that issue again.

Thanks for reporting and thanks for exercising the library!  I'll see if I can remedy this in the next few weeks if time permits but do not hesitate to let us know if you find a solution.  Thanks again!

-Mark

mcsauder, 2026-10-07:
I believe this was fixed with the addition of gtest 1.17 and C++17 in commits 50e0f44 and c1caa91.  Please reopen if this issue remains.

closed 2026-10-07 by mcsauder