crayzeewulf/libserial #184: macOS compile issue error 'linux/serial.h' file not found

thewoz, 2023-05-30:
Hi,
I am trying to compile the library from the master branch on macOS Ventura and I am getting the following error

`SerialPort.cpp:40:10: fatal error: 'linux/serial.h' file not found
`

 Reading #143 #147 I had understood that it had been solved

mcsauder, 2023-05-30:
@thewoz , try encapsulating that line as well:

```
        // @NOTE - termios.c_line is not a standard element of the termios
        // structure, (as per the Single Unix Specification 3).
        #ifdef __linux__
          port_settings.c_line = '\0' ;
        #endif
```

thewoz, 2023-05-30:
@mcsauder 

Leaving aside the warnings I have the following error:

/Users/thewoz/Downloads/libserial-master/src/SerialStream.cpp:921:31: fatal error: no member named 'GetAvailableSerialPorts' in 'LibSerial::SerialStreamBuf'
            return my_buffer->GetAvailableSerialPorts() ;

mcsauder, 2023-05-30:
Hi @thewoz , fetch the latest and checkout this branch: https://github.com/crayzeewulf/libserial/tree/linux_encapsulation

Once we get it working we can make a PR and get it merged in.

mcsauder, 2023-05-30:
Also, seeing that I can compile without `linux/serial.h` included in the SerialStreamBuf class, I've made another branch for you to try out if you wouldn't mind: https://github.com/crayzeewulf/libserial/tree/remove_linux_encapsulation

thewoz, 2023-05-31:
Hi @mcsauder sorry for the delay.

So:
in the branch remove_linux_encapsulation I get:
```
/ibserial-remove_linux_encapsulation/src/SerialPort.cpp:40:10: fatal error: 'linux/serial.h' file not found
```
while in the  branch linux_encapsulation I get
```
 /libserial-linux_encapsulation/src/SerialStream.cpp:938:7: fatal error: expected '}'
 #endif
```

in this case I corrected the error and recompiled it at the end I get 
```
./compile.sh: line 11: sphinx-build: command not found
```

mcsauder, 2023-05-31:
New code pushed up.  Thanks for testing this. I don't have access to your hardware.

Try this for sphinx: 
```
apt-get install python3-sphinx
```

mcsauder, 2023-05-31:
Or, rather, try this for sphinx in macOS:
```
sudo port install py38-sphinx
```

thewoz, 2023-05-31:
on remove_linux_encapsulation I get:
```
libserial-remove_linux_encapsulation/src/SerialPort.cpp:700:17: fatal error: out-of-line definition of 'GetAvailableSerialPorts' does not match any declaration in 'LibSerial::SerialPort'
    SerialPort::GetAvailableSerialPorts() const
```

while in the branch linux_encapsulation after have installed py38-sphinx I get:

```
+ sphinx-build -b html docs/user_manual/ docs/html/
Running Sphinx v6.2.1
WARNING: Invalid configuration value found: 'language = None'. Update your configuration to a valid language code. Falling back to 'en' (English).
making output directory... done

Theme error:
no theme named 'sphinx_rtd_theme' found (missing theme.conf?)
```

mcsauder, 2023-06-05:
Hi @thewoz , I pushed up a few fixes last week but didn't let you know.  Would you be willing to try the latest in those branches?

mcsauder, 2023-06-05:
Also, for the sphinx error, run `pip install sphinx-rtd-theme`

thewoz, 2023-06-06:
Hi @mcsauder 
on the "libserial-remove_linux_encapsulation" branch I get:
```
libserial-remove_linux_encapsulation/src/SerialPort.cpp:700:17: fatal error: out-of-line definition of 'GetAvailableSerialPorts' does not match any declaration in 'LibSerial::SerialPort'
    SerialPort::GetAvailableSerialPorts() const
```

while in the "libserial-linux_encapsulation" after have installed sphinx-rtd-theme everything is fine.

GravisZro, 2023-10-08:
This issue can be closed as it's been resolved.

closed 2023-10-09 by mcsauder